meshtastic/firmware #4662: Additional accelerometers (code hygiene)

dhskinner, 2024-09-09:
### Discussed in https://github.com/meshtastic/firmware/discussions/4614

<div type='discussions-op-text'>

<sup>Originally posted by **dhskinner** September  3, 2024</sup>
Hi - currently there a couple of accelerometers and motion sensors supported by Meshtastic that are all in one file at [.../src/AccelerometerThread.h](https://github.com/meshtastic/firmware/blob/master/src/AccelerometerThread.h) which makes it tricky to add new ones

Can I suggest from a code hygiene perspective it could be the right time to move AccelerometerThread.h into its own folder and tease apart the code into individual classes per sensor - like:

```
/src
    |->motion
        |-> AccelerometerThread.h
        |->MPU6050.cpp
        |->MPU6050.h
        |->...etc...
```

The reason for raising this is I'll shortly add a PR to add the ICM-20948 (n.b. the newer big brother from the MPU-6050) but it will need a couple of new code and header files to fully enable the onboard 9-Axis digital motion processor which is quite a cool... it would be a bit odd to add a bunch of new files into the [.../src/](https://github.com/meshtastic/firmware/tree/master/src) directory

Am more than happy to do the work and submit a small PR first for some light refactoring, but.... before I start I wanted to check in - will this annoy our admins by adding new folders and moving a bit of code around or is it do-able?

[edit - as of today (Sep 3rd) I did a scan of all open PR's and there doesn't seem to be any I could spot where anyone else is making changes to AccelerometerThread.h in parallel]
</div>

thebentern, 2024-09-09:
Looks like a good organizational strategy compared to what we have now. Excited to get more IMU and accelerometer support going

dhskinner, 2024-09-09:
> Have you had a look at how the temperature and other sensors are broken out at modules/telemetry/sensor? Is that the kind of approach you're thinking of following?

Yep am thinking pretty much the same approach as for  TelemetrySensor and  VoltageSensor 

> Looks like a good organizational strategy compared to what we have now. Excited to get more IMU and accelerometer support going

OK I'll do the refactoring and submit a PR in next few days. Currently I only have one of the existing accelerometers to test against (an MPU6050) so will see how it goes and might need to ask a favour for some help to test and confirm everything works as expected when its done. Much appreciated!

gjelsoe, 2024-09-09:
I've just added a new sensor (STK8BA53) to the code for the Radiomaster Bandit but haven't made a PR yet.

Some of the code are within AccelerometerThread.h, there are also 2 new files but added as a library link in platformio.ini

dhskinner, 2024-09-09:
> I've just added a new sensor (STK8BA53) to the code for the Radiomaster Bandit but haven't made a PR yet.

Ok I've put an initial refactored AccelerometerThread up here -> https://github.com/rocketflight/meshtastic_firmware/tree/ICM20948_Motion_Processor/src/motion

@gjelsoe this is all completely untested at the moment, so will take a day or three to become a PR and then a while longer to get some decent testing and everyone to be happy (I hope)... / I'll just fit in with whatever's happened in the meantime

gjelsoe, 2024-09-09:
@dhskinner I'll do my PR then.

dhskinner, 2024-09-11:
@gjelsoe Great - just took a skim through [#4667](https://github.com/meshtastic/firmware/pull/4667) and will wait a couple of days until that's merged before submitting a new PR to refactor (and in the meantime will finish off adding ICM-20948 also) - thanks!

closed 2024-09-28 by dhskinner