meshtastic/firmware #4664: [Bug]: SimpleJSON is unmaintained

fifieldt, 2024-09-09:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.0

### Description

We use code from the SimpleJSON library: https://github.com/MJPA/SimpleJSON
Specifically,
* src/serialization/JSON{.cpp,h}
* src/serialization/JSONValue{.cpp,h}

This code has not been updated in 8 years and the project appears unmaintained.

We should consider courses of action such as:
1. Investigate potential alternate libraries
2. Determine whether the meshtastic community should take on a maintainer role for this code
3. Patch some of the identified flaws

caveman99, 2024-09-09:
This is already the 2nd JSON lib we are using. Obvious choice would be the well maintained ArduinoJSON but is is rather big for the task.

https://github.com/meshtastic/json11

Suggestions welcome.

jp-bennett, 2024-09-09:
I vote for adding an MQTT bridge to the Meshtastic portfolio, and removing the JSON feature from the firmware.

fifieldt, 2024-09-15:
Probably wouldn't be hard to extend https://github.com/fifieldt/meshtastic-map-backend

to bridge MQTT and JSON in general, rather than just for maps. Could maybe even add some native homeassistant json stuff to make that integration easier...

@pdxlocations also has some nice MQTT bridge code.

garthvh, 2024-09-19:
> This is already the 2nd JSON lib we are using. Obvious choice would be the well maintained ArduinoJSON but is is rather big for the task.
> 
> https://github.com/meshtastic/json11
> 
> Suggestions welcome.

This is also abandoned by dropbox (maybe even older than our current library)

fifieldt, 2024-09-22:
this is what it looks like with ArduinoJson: https://github.com/meshtastic/firmware/pull/4661

fifieldt, 2024-09-22:
Has anyone looked into the protobuf library for this conversion? https://github.com/protocolbuffers/protobuf/blob/main/src/google/protobuf/util/json_util.h

garthvh, 2024-12-17:
JSON functionality in the firmware is in the process of being retired https://github.com/meshtastic/firmware/issues/5507