meshtastic/firmware #4687: [Feature Request]: Is there an option on/off bluetooth with q10 bb?

sim1988son, 2024-09-11:
### Platform

ESP32

### Description

I'm searching through the entire program/code and I can't find the on/off bluetooth option from the device position?

todd-herbert, 2024-10-07:
I think #4977 does a lot of the ground-work for this, but I think it probably only adds the hotkey for the M5Stack CardKB. 
It looks like Q10 BB / T-Deck keyboard would each need a few lines of code added to `input/kbI2cBase.cpp` (and someone to test).

closed 2024-12-15 by github-actions[bot]

hayschan, 2024-12-30:
Is the open source hardware Blackberry keyboard adapter boar [solderparty/pmod_bbq10_keyboard](https://github.com/solderparty/pmod_bbq10_keyboard) supported by Meshtastic now?

[image]