meshtastic/firmware #4703: [Feature Request]: Add a Node/Node user status to one of the periodic broadcast packets

Nestpebble, 2024-09-13:
### Platform

Cross-Platform

### Description

Nodes would benefit from a "status" in the same way that other chat platforms have a user status, e.g. Available, Away, Busy, etc.

However, Meshtastic nodes can be in many more states than just the user, e.g. Currently Unattended, Totally Unattended, Autoresponder only, Answering Machine only, Store & Forward Server, etc.

Adding a suitable set of statuses to a node would also improve the out-of-the-box experience for new users, who would be able to see that although there is a mesh around them, none are attended at the moment.

As more complex status messages are involved, bandwidth consumption increases and so do basic concerns like security and passive surveillance. However, with a few carefully selected statuses, I think this might be a valuable addition to the overall system.

thebentern, 2024-09-14:
This is a really interesting idea. I suppose it would live on User and could exist as a somewhat computed state at points, for instance receiving a text message would transition from a node we previously had as Away to Available, in addition to the user setting things explicitly

caveman99, 2024-09-14:
please make that a bitfield, that way a node can have more than one status. :-)

Nestpebble, 2024-09-15:
I've just thought this could also be a place to put stationary/mobile, or at least permanent station/mobile resting/mobile moving.