meshtastic/firmware #4706: [Bug]: T114 LED definitions a bit off

jhps, 2024-09-14:
### Category

Other

### Hardware

Heltec Mesh Node T114

### Firmware Version

2.5.0.d6dac17

### Description

"variant.h" for T114 define 3 controllable LEDs, whereas only one green one exists.
Two SK6812 NEOPIXEL LEDS are not defined.
(from https://resource.heltec.cn/download/Mesh%20Node%20T114/schematic_diagram.pdf)

The following hack still defines a non-existent blue LED, for the bluefruit library.
The Ambient Lighting module seems to do something with the NEOPIXELs.

diff --git a/variants/heltec_mesh_node_t114/variant.cpp b/variants/heltec_mesh_node_t114/variant.cpp
index cae079b7..b9b6ba74 100644
--- a/variants/heltec_mesh_node_t114/variant.cpp
+++ b/variants/heltec_mesh_node_t114/variant.cpp
@@ -35,10 +35,4 @@ void initVariant()
     // LED1 & LED2
     pinMode(PIN_LED1, OUTPUT);
     ledOff(PIN_LED1);
-
-    pinMode(PIN_LED2, OUTPUT);
-    ledOff(PIN_LED2);
-
-    pinMode(PIN_LED3, OUTPUT);
-    ledOff(PIN_LED3);
 }
diff --git a/variants/heltec_mesh_node_t114/variant.h b/variants/heltec_mesh_node_t114/variant.h
index e8c30599..a9871061 100644
--- a/variants/heltec_mesh_node_t114/variant.h
+++ b/variants/heltec_mesh_node_t114/variant.h
@@ -67,20 +67,20 @@ extern "C" {
 #define NUM_ANALOG_OUTPUTS (0)
 
 // LEDs
-#define PIN_LED1 (32 + 3) // 13 red (confirmed on 1.0 board)
-// Unused(by firmware) LEDs:
-#define PIN_LED2 (1 + 1)  // 14 blue
-#define PIN_LED3 (1 + 11) // 15 green
+#define PIN_LED1 (32 + 3) // 13 green (confirmed on 1.0 board)
 
-#define LED_RED PIN_LED3
-#define LED_BLUE PIN_LED1
-#define LED_GREEN PIN_LED2
+#define LED_BLUE PIN_LED1      // fake for bluefruit library
+#define LED_GREEN PIN_LED1
 
-#define LED_BUILTIN LED_BLUE
-#define LED_CONN PIN_GREEN
+#define LED_BUILTIN LED_GREEN
 
 #define LED_STATE_ON 0 // State when LED is lit
 
+#define HAS_NEOPIXEL                         // Enable the use of neopixels
+#define NEOPIXEL_COUNT 2                     // How many neopixels are connected
+#define NEOPIXEL_DATA 14                     // gpio pin used to send data to the neopixels
+#define NEOPIXEL_TYPE (NEO_GRB + NEO_KHZ800) // type of neopixels in use
+
 /*
  * Buttons
  */
@@ -206,4 +206,4 @@ No longer populated on PCB
  *        Arduino objects - C++ only
  *----------------------------------------------------------------------------*/
 
-#endif
\ No newline at end of file
+#endif

caveman99, 2024-09-14:
care to convert that to a regular pull request for review?

jhps, 2024-09-14:
> care to convert that to a regular pull request for review?

Could do, but what is to be done about the fake blue LED? Leave as aliased to GREEN?

PR #4710 (2024-09-17): Update T114 LED definitions to include only one simple controllable LED and two NEOPIXELs.
Fixes #4706.

An alias LED_BLUE for LED_GREEN/PIN_LED1 is maintained to keep bluefruit happy.

For reference: https://resource.heltec.cn/download/Mesh%20Node%20T114/schematic_diagram.pdf
closed 2024-09-17 by thebentern