meshtastic/firmware #4708: [Bug]: Firmware 2.5 Alpha - Admin channel "NO_Response"

thatSFguy, 2024-09-14:
### Category

Other

### Hardware

T-Lora v2 1.6, Rak4631

### Firmware Version

2.5.0.9ac0e26

### Description

Installed the firmware (2.5.0.9ac0e26) on both Tlora and RAK devices.  Admin channels were already configured on the devices.  

1. Tested via Android the remote configuration from Tlora to RAK, RAK to Tlora, Tlora to Tlora.  Get the "NO RESPONSE" error.
2. Utilized a portuino  (firmware 2.4) to attempt CLI configuration of 2.5 nodes and received the "NO Response" error
3. Regression tested admin configuration from 2.5 node to the portunino 2.4 node and that worked

---------------------
Other information:

- Direct Message sending from RAK to Tlora didn't work
- Direct Message sending from Tlora to Tlora did work.
- sending messages on the admin channel are received by all notes

caveman99, 2024-09-14:
what version of the android app do you have? you need at least 2.4.5

thatSFguy, 2024-09-14:
I'm on 2.4.5 (Beta) via the app store.

thatSFguy, 2024-09-14:
updated bug with logs from tlora to rak:

1. Admin request from Android
2. DM message

thatSFguy, 2024-09-14:
Here is the log from the Tlora Serial monitor when attempting to send DM from RAK to TLora:

WARN | 18:43:57 1412 [Router] No suitable channel found for decoding, hash was 0x0!

DEBUG | 18:43:57 1412 [Router] packet decoding failed or skipped (no PSK?) (id=0xa367996f fr=0x08 to=0xa4, WantAck=1, HopLim=7 Ch=0x0 encrypted rxtime=1726339437 rxSNR=11 rxRSSI=-34 hopStar

DEBUG | 18:43:57 1412 [Router] Module 'routing' wantsPacket=1

INFO | 18:43:57 1412 [Router] Received routing from=0xb2982108, id=0xa367996f, portnum=-228327395, payloadlen=6970

DEBUG | 18:43:57 1412 [Router] Routing sniffing (id=0xa367996f fr=0x08 to=0xa4, WantAck=1, HopLim=7 Ch=0x0 encrypted rxtime=1726339437 rxSNR=11 rxRSSI=-34 hopStart=7)

DEBUG | 18:43:58 1412 [Router] Partially randomized packet id 95162455

WARN | 18:43:58 1412 [Router] Alloc an err=6,to=0xb2982108,idFrom=0xa367996f,id=0x5ac1057

DEBUG | 18:43:58 1412 [Router] localSend to channel 1

DEBUG | 18:43:58 1412 [Router] Add packet record (id=0x05ac1057 fr=0xa4 to=0x08, WantAck=0, HopLim=2 Ch=0x1 Portnum=5 requestId=a367996f rxtime=1726339438 priority=120)

DEBUG | 18:43:58 1412 [Router] Expanding short PSK #1

DEBUG | 18:43:58 1412 [Router] Using AES128 key!

DEBUG | 18:43:58 1412 [Router] enqueuing for send (id=0x05ac1057 fr=0xa4 to=0x08, WantAck=0, HopLim=2 Ch=0x8 encrypted rxtime=1726339438 hopStart=2 priority=120)

DEBUG | 18:43:58 1412 [Router] txGood=9,rxGood=61,rxBad=0

DEBUG | 18:43:58 1412 [Router] Delivering rx packet (id=0xa367996f fr=0x08 to=0xa4, WantAck=1, HopLim=7 Ch=0x0 encrypted rxtime=1726339437 rxSNR=11 rxRSSI=-34 hopStart=7)

DEBUG | 18:43:58 1412 [Router] Forwarding to phone (id=0xa367996f fr=0x08 to=0xa4, WantAck=1, HopLim=7 Ch=0x0 encrypted rxtime=1726339437 rxSNR=11 rxRSSI=-34 hopStart=7)

DEBUG | 18:43:58 1412 [Router] Attempting PKI decryption

INFO | 18:43:58 1412 [Router] Random nonce value: 297.....

DEBUG | 18:43:58 1412 [Router] DH Output: : 7b 7d 4b.......................

DEBUG | 18:43:58 1412 [Router] Attempting decrypt using nonce: : 6f 99 67 a3........... 

DEBUG | 18:43:58 1412 [Router] Attempting decrypt using shared_key: : 97 f8 ...............

WARN | 18:43:58 1413 [Router] No suitable channel found for decoding, hash was 0x0!

DEBUG | 18:43:58 1413 [Router] Module 'routing' considered

WARN | 18:43:58 1413 [RadioIf] Can not send yet, busyRx

WARN | 18:43:58 1413 [RadioIf] Can not send yet, busyRx

WARN | 18:43:58 1413 [RadioIf] Can not send yet, busyRx

thatSFguy, 2024-09-14:
Updated the bug and removed the DM issue, so the focus is only remote administration.  The DM issue seems to be related an "unsupported crystal" on the TLora device.  I'll open a separate bug for that.

pyro2927, 2024-09-15:
I'm seeing the same issue on a Heltec v3 and Lilygo T-Deck. Firmware version `2.5.0.9ac0e26`, accessing via the Python CLI. The Heltec v3 is on WiFi.

```
$ meshtastic --host 192.168.1.25 --dest '!bd4c2170' --get device  
Connected to radio
Requesting current config from remote node (this can take a while).
Error on response: NO_RESPONSE
Completed getting preferences
```

jp-bennett, 2024-09-15:
An unintended consequence of all the 2.5 changes is that the legacy admin channel won't work at all between 2.5 devices. Using your public key as the admin key on the device you want to control is the way to go.

Controlling a 2.5 device from a 2.4 node might not work either, as the 2.4 node doesn't have the updated protobuf fields to handle the session key. I need to test that case out myself.

pyro2927, 2024-09-15:
Both of my devices are on 2.5, so I don't think that would be the issue. That said, I hadn't added the public key of my primary node as the admin key on the other. After doing that I run into a slightly different issue:

jp-bennett, 2024-09-15:
@pyro2927 can you successfully send DMs between the two nodes? If not, you probably have a key mismatch from bouncing around your firmware versions. Use the Python client and --nodes to compare the actual public key to what the opposite device sees in its NodeDB, and remove the node from the nodeDB if it doesn't match.

slawekgajzner, 2024-09-15:
Correct me if I'm wrong.

For the admin channel to work (not in legacy mode) the packet must also be encrypted (mp.pki_encrypted == true)

AdminModule.cpp:
if (!((mp.from == 0 && !config.security.is_managed) || messageIsResponse(r) ||
(strcasecmp(ch->settings.name, Channels::adminChannel) == 0 && config.security.admin_channel_enabled) ||
(**mp.pki_encrypted** && memcmp(mp.public_key.bytes, config.security.admin_key[0].bytes, 32) == 0))) {
LOG_INFO("Ignoring admin payload %i\n", r->which_payload_variant);
return handled; } 

For pki_encrypted to be true, the message must arrive on channel 0 

Router.cpp (perhapsDecode)
if (**p->channel == 0** && p->to == nodeDB->getNodeNum() && p->to > 0 && p->to != NODENUM_BROADCAST && nodeDB->getMeshNode(p->from) != nullptr && nodeDB->getMeshNode(p->from)->user.public _key.size > 0 && 
nodeDB->getMeshNode(p->to)->user.public_key.size > 0 && rawSize > 12)
{
 LOG_DEBUG("Attempting PKI decryption\n");

 if (crypto->decryptCurve25519(p->from, p->id, rawSize, ScratchEncrypted, bytes)) {
LOG_INFO("PKI Decryption worked!\n");
memset(&p->decoded, 0, sizeof(p->decoded));
rawSize -= 12;
if (pb_decode_from_bytes(bytes, rawSize, &meshtastic_Data_msg, &p->decoded) &&
p->decoded.portnum != meshtastic_PortNum_UNKNOWN_APP) {
decrypted = true;
LOG_INFO("Packet decrypted using PKI!\n");
**p->pki_encrypted = true;**

but at this point the channel is still hashed and probably won't be equal to 0, only later does the channel unhash.

jp-bennett, 2024-09-15:
> but at this point the channel is still hashed and probably won't be equal to 0, only later does the channel unhash.

PKI packets always set the hash to 0. That check is probably superfluous, but I don't think it would cause problems.

slawekgajzner, 2024-09-15:
What about is_licensed, if it is set to true then PKI is not used so the pki_encrypted flag will not be set,
 maybe that's why admin doesn't work. 
If I set is_licensed to true I get the No Response message. 
If I turn off is_licensed (License amateur radio) everything works 

Router.cpp 
if **(!owner.is_licensed** && config.security.private_key.size == 32 && p->to != NODENUM_BROADCAST && node != nullptr && node->user.public_key.size > 0 && p->decoded.port num != meshtastic_PortNum_TRACEROUTE_APP && p->decoded.portnum != meshtastic_PortNum_NODEINFO_APP && p->decoded.portnum != meshtastic_PortNum_ROUTING_APP && p->decoded.portnum != meshtastic_PortNum_POSITION_APP) 
{
...

jp-bennett, 2024-09-15:
> What about is_licensed

Ham mode means no encryption at all. Definitely precludes PKI for admin. Though using the legacy admin channel would likely work.

garthvh, 2024-09-15:
You should not be able to use the legacy admin channel with no encryption

pyro2927, 2024-09-15:
@jp-bennett looks like my T-Deck is experiencing [this issue](https://meshtastic.discourse.group/t/public-key-mismatch-on-first-interaction/14702), which is weird because I have confirmed the keys are not changing and are correct. Tried to send DMs and [had this problem](https://meshtastic.discourse.group/t/no-channel-error-while-direct-messaging-2-5-0/14533/3).

I switched to using two Heltec v3s (both on `2.5.0.9ac0e26`). Reset Node DBs on all three devices. Was able to get admin controls to work, but am still seeing weird errors of "Public Key Mismatch" intermittently.

jp-bennett, 2024-09-15:
> @jp-bennett looks like my T-Deck is experiencing [this issue](https://meshtastic.discourse.group/t/public-key-mismatch-on-first-interaction/14702), which is weird because I have confirmed the keys [...]

There seems to be a false positive in iOS, causing that key mismatch error. It's getting worked on.

thatSFguy, 2024-09-16:
Sorry, I'm caught up with personal stuff and work, so I've not had a chance to test with the CLI.  I did a fresh install with complete erase on all nodes to eliminate any potential legacy issues before opening this bug.

jp-bennett, 2024-09-17:
> Both of my devices are on 2.5, so I don't think that would be the issue. That said, I hadn't added the public key of my primary node as the admin key on the other. After doing that I run into a slightly different issue:
> 
> ```
> $ meshtastic --host 192.168.1.25 --dest '!bd4c2170' --get device
> Connected to radio
> Requesting current config from remote node (this can take a while).
> Aborting due to: Timed out waiting for an acknowledgment
> ```

Could you snag a serial output log on the node to be controlled, when this happens? This is a really interesting case.

thatSFguy, 2024-09-18:
here is the output of a admin request from Android 2.4.5.  Nodes are running 2.5.1 .  Log is from the reeiving node.

```Shell DEBUG | 02:27:12 181 [Power] Battery: usbPower=1, isCharging=1, batMv=4225, batPct=100
DEBUG | 02:27:13 181 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=26, time 428 ms
DEBUG | 02:27:13 181 [RadioIf] Lora RX (id=0x0695114f fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0xd2 encrypted rxSNR=12.5 rxRSSI=-27 hopStart=7)
DEBUG | 02:27:13 181 [RadioIf] AirTime - Packet received : 428ms
DEBUG | 02:27:13 181 [Router] Add packet record (id=0x0695114f fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0xd2 encrypted rxSNR=12.5 rxRSSI=-27 hopStart=7)
DEBUG | 02:27:13 181 [Router] Using channel 1 (hash 0xd2)
DEBUG | 02:27:13 181 [Router] Using AES256 key!
DEBUG | 02:27:13 181 [Router] decoded message (id=0x0695114f fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0x1 Portnum=6 WANTRESP rxtime=1726626433 rxSNR=12.5 rxRSSI=-27 hopStart=7)
DEBUG | 02:27:13 181 [Router] handleReceived(REMOTE) (id=0x0695114f fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0x1 Portnum=6 WANTRESP rxtime=1726626433 rxSNR=12.5 rxRSSI=-27 hopStart=7)
DEBUG | 02:27:13 181 [Router] Module 'Admin' wantsPacket=1
INFO  | 02:27:13 181 [Router] Received Admin from=0x1c2cbfd8, id=0x695114f, portnum=6, payloadlen=2
INFO  | 02:27:13 181 [Router] Handling admin payload 1
INFO  | 02:27:13 181 [Router] Client is getting channel 0
DEBUG | 02:27:13 181 [Router] Setting admin key to : 2d cf 46 29 04 b4 78 d8 
DEBUG | 02:27:13 181 [Router] Partially randomized packet id 1360079679
INFO  | 02:27:13 181 [Router] Asked module 'Admin' to send a response
DEBUG | 02:27:13 181 [Router] Module 'routing' wantsPacket=1
INFO  | 02:27:13 181 [Router] Received routing from=0x1c2cbfd8, id=0x695114f, portnum=6, payloadlen=2
DEBUG | 02:27:13 181 [Router] Routing sniffing (id=0x0695114f fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0x1 Portnum=6 WANTRESP rxtime=1726626433 rxSNR=12.5 rxRSSI=-27 hopStart=7)
DEBUG | 02:27:13 181 [Router] Some other module has replied to this message, no need for a 2nd ack
DEBUG | 02:27:13 181 [Router] Delivering rx packet (id=0x0695114f fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0x1 Portnum=6 WANTRESP rxtime=1726626433 rxSNR=12.5 rxRSSI=-27 hopStart=7)
DEBUG | 02:27:13 181 [Router] Update DB node 0x1c2cbfd8, rx_time=1726626433
DEBUG | 02:27:13 181 [Router] Forwarding to phone (id=0x0695114f fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0x1 Portnum=6 WANTRESP rxtime=1726626433 rxSNR=12.5 rxRSSI=-27 hopStart=7)
DEBUG | 02:27:13 181 [Router] Module 'routing' considered
DEBUG | 02:27:13 182 [Router] Sending response (id=0x51112b3f fr=0xa4 to=0xd8, WantAck=1, HopLim=2 Ch=0x1 Portnum=6 requestId=695114f rxtime=1726626433 priority=70)
DEBUG | 02:27:13 182 [Router] Update DB node 0x25c0dba4, rx_time=1726626433
DEBUG | 02:27:13 182 [Router] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=47, time 583 ms
DEBUG | 02:27:13 182 [Router] Setting next retransmission in 8284 msecs:  (id=0x51112b3f fr=0xa4 to=0xd8, WantAck=1, HopLim=2 Ch=0x1 Portnum=6 requestId=695114f rxtime=1726626433 priority=70)
DEBUG | 02:27:13 182 [Router] Add packet record (id=0x51112b3f fr=0xa4 to=0xd8, WantAck=1, HopLim=2 Ch=0x1 Portnum=6 requestId=695114f rxtime=1726626433 priority=70)
DEBUG | 02:27:13 182 [Router] Using AES256 key!
DEBUG | 02:27:13 182 [Router] enqueuing for send (id=0x51112b3f fr=0xa4 to=0xd8, WantAck=1, HopLim=2 Ch=0xd2 encrypted rxtime=1726626433 hopStart=2 priority=70)
DEBUG | 02:27:13 182 [Router] txGood=3,rxGood=8,rxBad=0
WARN  | 02:27:13 182 [RadioIf] Can not send yet, busyRx
DEBUG | 02:27:13 182 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=26, time 428 ms
DEBUG | 02:27:13 182 [RadioIf] Lora RX (id=0x06951150 fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0xd2 encrypted rxSNR=12 rxRSSI=-26 hopStart=7)
DEBUG | 02:27:13 182 [RadioIf] AirTime - Packet received : 428ms
DEBUG | 02:27:13 182 [Router] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=26, time 428 ms
DEBUG | 02:27:13 182 [Router] Add packet record (id=0x06951150 fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0xd2 encrypted rxSNR=12 rxRSSI=-26 hopStart=7)
DEBUG | 02:27:13 182 [Router] Using channel 1 (hash 0xd2)
DEBUG | 02:27:13 182 [Router] Using AES256 key!
DEBUG | 02:27:13 182 [Router] decoded message (id=0x06951150 fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0x1 Portnum=6 WANTRESP rxtime=1726626433 rxSNR=12 rxRSSI=-26 hopStart=7)
DEBUG | 02:27:13 182 [Router] handleReceived(REMOTE) (id=0x06951150 fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0x1 Portnum=6 WANTRESP rxtime=1726626433 rxSNR=12 rxRSSI=-26 hopStart=7)
DEBUG | 02:27:13 182 [Router] Module 'Admin' wantsPacket=1
INFO  | 02:27:13 182 [Router] Received Admin from=0x1c2cbfd8, id=0x6951150, portnum=6, payloadlen=2
INFO  | 02:27:13 182 [Router] Ignoring admin payload 5
INFO  | 02:27:13 182 [Router] Asked module 'Admin' to send a response
DEBUG | 02:27:13 182 [Router] Module 'routing' wantsPacket=1
INFO  | 02:27:13 182 [Router] Received routing from=0x1c2cbfd8, id=0x6951150, portnum=6, payloadlen=2
DEBUG | 02:27:13 182 [Router] Routing sniffing (id=0x06951150 fr=0xd8 to=0xa4, WantAck=1, HopLim=7 Ch=0x1 Portnum=6 WANTRESP rxtime=1726626433 rxSNR=12 rxRSSI=-26 hopStart=7)
DEBUG | 02:27:13 182 [Router] Partially randomized packet id 3549609792
DEBUG | 02:27:13 182 [Router] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=27, time 436 ms
[... cut ...]
DEBUG | 02:27:27 195 [Router] Add packet record (id=0x286c63d5 fr=0xd8 to=0xa4, WantAck=0, HopLim=1 Ch=0xd2 encrypted rxSNR=11 rxRSSI=-61 hopStart=2)
DEBUG | 02:27:27 195 [Router] Ignoring incoming msg we've already seen (id=0x286c63d5 fr=0xd8 to=0xa4, WantAck=0, HopLim=1 Ch=0xd2 encrypted rxSNR=11 rxRSSI=-61 hopStart=2)
DEBUG | 02:27:27 195 [Router] cancelSending id=0x286c63d5, removed=0
DEBUG | 02:27:27 195 [Router] Incoming message was filtered from 0x1c2cbfd8
DEBUG | 02:27:32 201 [Power] Battery: usbPower=1, isCharging=1, batMv=4225, batPct=100
 *  Terminal will be reused by tasks, press any key to close it.   ```

jp-bennett, 2024-09-18:
> here is the output of a admin request from Android 2.4.8. Nodes are running 2.5.1 . Log is from the receiving node. DM was working prior to the admin request.

This looks like it's using a channel named "admin" rather than public key messages.

And it looks like it should have worked, too. I can see the responsea go back. Did the Android app just time out?

thatSFguy, 2024-09-18:
The error I get is NO_RESPONSE.  Assuming that it's actually a timeout.

thatSFguy, 2024-09-25:
It's been a while now and I have all my nodes upgraded to 2.5.  I found that biting the bullet and just upgrading all the nodes and using the Security/ Admin PKI seems to work fine.  Mixing and matching by trying to use the legacy admin channel is just problematic. 

I do have one problem with my portduino.  Somehow the web interface let me set the "managed mode" which was fine when I had the admin channel.   Well, I deleted the admin channel and now I can't change any settings.  Any recommendations on how to reset it to factory,  since it won't let me do it via cli or web. Thanks!

thatSFguy, 2024-09-25:
To all of those that come after me.  Here are some recommendations:

1. If you upgrade to 2.5, upgrade all nodes to avoid remote admin problems 
2. The "Managed Remotely" setting in the security setting should be off.  If it is on, and you remove the admin channel without updating the Remote admin key your node is locked from all config changes.
3. Make sure you update your Admin key in the Security section by adding the Public key of the controlling node (see screeshot) to the remote admin node.
4. Test your new administration, once working you can remove the legacy admin channel.
5. If you locked yourself out of the node with the Remotely Managed feature you will just have to do a full erase/install.  On the portduino I had to downgrade to 2.4, change the settings under device section and the upgrade again.
[image]

I hope my experience helps someone else!  Some of my issue are probably just the lack of understanding of the new features! 

Thanks @jp-bennett fir all your help!

thatSFguy, 2024-09-25:
Closing this bug.  If needed I can open a feature request to enhance validation on "Remotely Managed toggle" to prevent deleting the legacy admin channel if no admin key has been provided.

closed 2024-09-25 by thatSFguy