meshtastic/firmware #4711: [Bug]: Firmware 2.5 Alpha - DM message does not work from a one TLORA Node

thatSFguy, 2024-09-14:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.5.0.9ac0e26

### Description

By chance, the first node I upgraded had a problem sending DM messages.  Regular messages work fine. One time I saw a unsupported crystal message in the serial console when I did the update, but not when I tried to reproduce it.  I'm not sure if this is an edge case or how many people are using Alpha, but I figured I'd log this one if others experience an issue with it.  

The end user experience is that when you try to send a DM you just the crossed out cloud.

### Relevant log output

```Shell
ERROR | ??:??:?? 45 [Router] Can't decode protobuf reason='end-of-stream', pb_msgdesc 0x3f40ccc0
ERROR | ??:??:?? 45 [Router] Invalid protobufs in received mesh packet id=0x97219ab4 (bad psk?)!
ERROR | ??:??:?? 45 [Router] Can't decode protobuf reason='end-of-stream', pb_msgdesc 0x3f40ccc0
ERROR | ??:??:?? 45 [Router] Invalid protobufs in received mesh packet id=0x97219ab4 (bad psk?)!
ERROR | ??:??:?? 136 [Router] Invalid portnum (bad psk?)!
ERROR | ??:??:?? 139 [Router] Invalid portnum (bad psk?)!
ERROR | ??:??:?? 139 [Router] Invalid portnum (bad psk?)!
ERROR | ??:??:?? 136 [Router] Invalid portnum (bad psk?)!
ERROR | ??:??:?? 139 [Router] Invalid portnum (bad psk?)!
ERROR | ??:??:?? 139 [Router] Invalid portnum (bad psk?)!
```

jp-bennett, 2024-09-15:
Use the Python client with --nodes and see if the nodes are properly capturing each other's public keys. There's been a few reports that things start working better after a factory reset. See if you can catch a serial snippet from both sides of the failing DM, as that has some useful debug info.

thatSFguy, 2024-09-16:
I'm not sure how to tell based on the output if it's capturing the public key.  I've seen it in the logs, but not with --nodes.   However, today the one node that failed 100% of the time "Blue Node" just started working.  As in my other report, all nodes were flashed w/ erase and configured from scratch.  I did that several times with the "blue node' since I was having issues.  I will say that the node has been powered down for about 48 hours.  Not sure if there was something residual left behind.   

  N │ User                    │ ID        │ AKA   │ Hardware       │ Latitude   │ Longitude   │ Altitude   │ Battery   │ Channel util.   │ Tx air util.   │ SNR      │ Hops Away   │   Channel │ LastHeard           │ Since       │
╞═════╪═════════════════════════╪═══════════╪═══════╪════════════════╪════════════╪═════════════╪════════════╪═══════════╪═════════════════╪════════════════╪══════════╪═════════════╪═══════════╪═════════════════════╪═════════════╡
│   1 │ Blue Node               │ !25c0dba4 │ Blue  │ TLORA_V2_1_1P6 │ 43.1590°   │ -85.6463°   │ 260 m      │ 101%      │ 8.11%           │ 0.04%          │ N/A      │ 0/unknown   │         0 │ 2024-09-16 18:43:13 │ 1 sec ago   │
├─────┼─────────────────────────┼───────────┼───────┼────────────────┼────────────┼─────────────┼────────────┼───────────┼─────────────────┼────────────────┼──────────┼─────────────┼───────────┼─────────────────────┼─────────────┤
│   2 │ Black GPS               │ !25d63230 │ BLKG  │ TLORA_V2_1_1P6 │ 43.1620°   │ -85.6556°   │ 260 m      │ N/A       │ N/A             │ N/A            │ 13.25 dB │ 0/unknown   │         0 │ 2024-09-16 18:42:54 │ 20 secs ago │
├─────┼─────────────────────────┼───────────┼───────┼────────────────┼────────────┼─────────────┼────────────┼───────────┼─────────────────┼────────────────┼──────────┼─────────────┼───────────┼─────────────────────┼─────────────┤

thatSFguy, 2024-09-16:
O.k. I fired up another 2.5 node and while I can see it in the list, I'm unable to send DM's either way.  Maybe this is more related to the other Bug I opened about the randomness of DM's.  Since I can't reproduce the exact issue with the "Blue Node"  I'll close this bug.

closed 2024-09-16 by thatSFguy

jp-bennett, 2024-09-17:
> I'm not sure how to tell based on the output if it's capturing the public key. I've seen it in the logs, but not with --nodes.
Looks like you need to update your meshtastic python client to 2.5 to get the extra output. `pip install --upgrade meshtastic` if you're using pip to manage your install.