meshtastic/firmware #4714: [Bug]: Heltec T114 HW_VENDOR is set to generic nrf52840dk

dahanc, 2024-09-15:
### Category

Hardware Compatibility

### Hardware

Heltec Mesh Node T114

### Firmware Version

2.5.0.9ac0e26

### Description

The app shows Heltec T114 nodes as "nrf52840dk" (such as in the node list page and on the device connection page: "Connected to radio (nrf52840dk 2.5.0.9ac0e26)"). It looks like this is because while there's a specific [HW_VENDOR definition](https://github.com/meshtastic/firmware/blob/8893529653cb23060f88f3ff56b092e5291facac/src/platform/esp32/architecture.h#L168) for the T114, it's in the platform/esp32/architecture.h file. Seems like it should be in [platform/nrf52/architecture.h](https://github.com/meshtastic/firmware/blob/8893529653cb23060f88f3ff56b092e5291facac/src/platform/nrf52/architecture.h) instead.

dahanc, 2024-09-15:
Also, https://github.com/meshtastic/firmware/blob/master/boards/heltec_mesh_node_t114.json has `"extra_flags": "-DARDUINO_NRF52840_PCA10056 -DNRF52840_XXAA",`. Should `-DARDUINO_NRF52840_PCA10056` be there?

thebentern, 2024-09-15:
Good catch. We must have missed this in the initial PR from Heltec.

PR #4719 (2024-09-16): Add Heltec T1114 hardware model to build
Closes #4714
PR #4772 (2024-09-19): Another missed define for the T114
HW_MODEL still isn't being set correctly. It looks like this change is needed to get the proper define. I flashed a firmware this change, and the model is finally showing as "heltec-mesh-node-t114" in the Android app.

fixes #4714
closed 2024-09-16 by thebentern