meshtastic/firmware #4718: [Bug]: The backlight stopped turning on when pressing the button

dmitrym-web, 2024-09-15:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.4.2.5b45303 Beta

### Description

After updating the firmware, the backlight stopped turning on when pressing the button, when updating the firmware to other versions, older or newer, the result is the same

todd-herbert, 2024-09-15:
You mean the capacitive touch button, yeah? My T-Echo never did this, but I've heard other people talk about this too. I think it maybe depends on which version of the bootloader your device shipped with(?)

Currently, you can toggle the backlight by pressing the user button four times. It's a lot of clicking, but it's the best compromise we could come up with.

There's some more info here:https://github.com/meshtastic/firmware/issues/3649#issuecomment-2060159524

dmitrym-web, 2024-09-15:
[image]

Yes, the P 0.11 button on the diagram. Four times is not convenient to press and this method does not work. On what firmware did this work?

closed 2024-09-15 by thebentern

todd-herbert, 2024-09-15:
> On what firmware did this work?

I believe it is related to the bootloader, not the firmware. For some users, this never worked.

> Four times is not convenient

I agree, but it is a compromise. The discussion took place here: https://github.com/meshtastic/firmware/issues/3533 and https://github.com/meshtastic/firmware/pull/3560.

It would certainly be convenient to have the option to use the capacitive touch button to control the backlight, but needs to be done in a way that respects users who have previously removed this button. It's something I do intend to explore, but longer-term, as part of a UI redesign.

dmitrym-web, 2024-09-20:
This was a bad decision, it could have been forked for those who don't use a store bought device.

todd-herbert, 2024-09-20:
I don't believe this has ever been a feature of the Meshtastic firmware. I believe some older devices may have had a special bootloader (separate from the firmware), installed by LilyGo which performed this feature, however modern T-Echos do not ship with this(?)

I *do* like the idea of using the capacitive touch button for the display light, and am hoping to add the feature to the firmware in future, in a way that also respects users who have removed it as part of customisation.

These users removed the touch button long ago, because for many T-Echos, it has never controlled the display light, from factory new, even before P0.11 was remapped.

todd-herbert, 2024-09-20:
This afternoon, I have taken the time to look back through old code, right back into v1.x.
I didn't see any version where P0.11 controls T-Echo's display light.
This seems to confirm the theory that is a feature of an old bootloader.

If you are keen to add this feature for yourself immediately:
* Set up to [build the firmware from source-code](https://meshtastic.org/docs/development/firmware/build/)
* Edit [ButtonThread.cpp](https://github.com/meshtastic/firmware/blob/0f3450ad441f766bd451e9f2d2cfea09b5730431/src/ButtonThread.cpp#L207-L218)
* Below Line 207 (`case BUTTON_EVENT_TOUCH_LONG_PRESSED`), insert the following code (untested):
```cpp
#if defined(USE_EINK) && defined(PIN_EINK_EN)
        digitalWrite(PIN_EINK_EN, digitalRead(PIN_EINK_EN) == LOW);
#endif
```

augustozanellato, 2024-09-20:
It probably wasn’t in the bootloader but in the custom Meshtastic version that lilygo ships preinstalled available [here](https://github.com/Xinyuan-LilyGO/T-Echo/blob/537b5233f01365d60d18e64964fa3a7306ea769a/firmware/factory/firmware-t-echo-2.0.10.e09b12c.uf2)

todd-herbert, 2024-09-20:
That would make more sense! I wasn't sure exactly how the bootloader could have impacted it: whether or not it was actually possible for code executed early on to have a persisting impact once the firmware loads.. I didn't realize that it was as simple as them shipping with a custom firmware.

todd-herbert, 2024-09-20:
> It probably wasn’t in the bootloader but in the custom Meshtastic version that lilygo ships preinstalled available [here](https://github.com/Xinyuan-LilyGO/T-Echo/blob/537b5233f01365d60d18e64964fa3a7306ea769a/firmware/factory/firmware-t-echo-2.0.10.e09b12c.uf2)

Out of interested, I compared the hashes, and this file on LilyGo's repo appears to be identical to the [official T-Echo firmware release for 2.0.10](https://github.com/meshtastic/firmware/releases/tag/v2.0.10.e09b12c)

I still does seem most likely that a custom firmware is responsible here, just maybe not this exact build?

dmitrym-web, 2024-09-20:
> I don't believe this has ever been a feature of the Meshtastic firmware. I'm not certain, but I believe some older devices may have had a special bootloader (separate from the firmware), installed by LilyGo which performed this feature, however modern T-Echos do not ship with this(?)
> 
> I _do_ like the idea of using the capacitive touch button for the display light, and am hoping to add the feature to the firmware in future, in a way that also respects users who have removed the touch button as part of customisation.
> 
> These users removed the touch button long ago, because for many T-Echos, it has never controlled the display light, from factory new, even before P0.11 was remapped. For a very long time, for many users, the touch button had no function.

I bought the device a week ago and the button worked and the firmware was messy, unfortunately, I did not remember the version, another person from my city bought it a week ago and his button works

dmitrym-web, 2024-09-20:
> Out of interest, I compared the hashes, and this file on LilyGo's repo appears to be identical to the [official T-Echo firmware release for 2.0.10](https://github.com/meshtastic/firmware/releases/tag/v2.0.10.e09b12c)
> 
> I feel like you're probably still right: a custom LilyGo-shipped firmware would make sense here, but maybe not this exact build?

Is it possible to transfer firmware from one device to another?