meshtastic/firmware #4723: [Bug]:

adingbatponder, 2024-09-15:
### Category

Other

### Hardware

Heltec Mesh Node T114

### Firmware Version

2.4.3.91d6612

### Description

Messages above 47 characters cannot be sent. Other are reporting that the message length that causes failure to send is variable  https://www.reddit.com/r/meshtastic/comments/1fhc47k/heltec_t114_message_sending_bugfault

adingbatponder, 2024-09-15:
https://meshtastic.discourse.group/t/problems-with-heltec-t114/14489

adingbatponder, 2024-09-15:
Does it forward packets normally or are certain length messages not forwarded?

GUVWAF, 2024-09-15:
Can you try setting a lower transmit power (e.g. 3 dBm) and see if the issue persists?

lupusworax, 2024-09-16:
lowering the transmit power  to 3 dbm did the trick! it transmitted for me from one T114 to another.

lupusworax, 2024-09-16:
[image]

willfootwill, 2024-09-16:
exact same behaviour here - ok below 8db

willfootwill, 2024-09-16:
I tried going back to earlier firmware 2.4.xxx, no improvement

lupusworax, 2024-09-16:
I am on 2.5 most recent release. If this issue is hardware based its bad, really bad!! because how I see it now every T114 so far is affected.

NeilMartin, 2024-09-16:
My T114 is running 2.4.2 and I just sent a 71 character message to LongFast. Received successfully on my T-Deck (2.4.2).

lupusworax, 2024-09-16:
Also tested to send from my T114 ( stock 27dbm)  to my Heltec V3 a much longer message then 71  successfully 3 times out of 4.

DTopp, 2024-09-17:
Today at 27dBm I can send around 60 character messages, up to 30 characters no problem after that I had to resend a couple of messages.

At 3dBm I'm managing to send 200 character messages (stopped there as I wasn't sure what the limit is) and they've been getting through no problem.

aljaxus, 2024-09-17:
can send 144-char (base64 encoded, and 107-char decoded)  messages with `tx_power` set to 5. (more info in [reddit reply](https://www.reddit.com/r/meshtastic/comments/1fhc47k/comment/lnk1m0j/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button))

thebentern, 2024-09-17:
Thinking out loud... Given this symptom combined with some of the sporadic BLE issues, I wonder if we're looking at some hardware level issues with the LDO converter not being able to supply enough current to the MCU + peripherals under certain loads.

lupusworax, 2024-09-17:
Would a big enough capacitor at the LDO be able to buffer that?

thebentern, 2024-09-17:
> Would a big enough capacitor at the LDO be able to buffer that?

Potentially. If someone is willing to sacrifice hardware modifications in the name of science, we can find out. 😅

fifieldt, 2024-09-17:
Q) Does everyone above have the version with a screen?

lupusworax, 2024-09-17:
Screen+ GPS  and Screen no GPS but lots of other added stuff like vibra, buzzer ( both fired by a mosfet feeded directly from the 3.3volt rail)  and a smd LED but that barely falls into weight. I have to retry with a unmodified T114 as well when I get home. ( With screen but no GPS)

lupusworax, 2024-09-17:
> > Would a big enough capacitor at the LDO be able to buffer that?
> 
> Potentially. If someone is willing to sacrifice hardware modifications in the name of science, we can find out. 😅

I think I will give it a try, what size capacitor should be right? The CE6260 puts out 500mA if I got the correctly?

NeilMartin, 2024-09-17:
> Q) Does everyone above have the version with a screen? GPS?
> Screen and GPS.

Screen and GPS for me too.

todd-herbert, 2024-09-17:
> Thinking out loud... Given this symptom combined with some of the sporadic BLE issues, I wonder if we're looking at some hardware level issues with the LDO converter not being able to supply enough current to the MCU + peripherals under certain loads.

I would have suspected the same thing, but I've got it on the scope right now though and I can't measure any significant dip in voltage when transmitting. This was with screen only. I'll connect the GPS and see if it changes.

fifieldt, 2024-09-18:
I've managed to reproduce on a T114 with a GPS and no screen. So the screen is not a factor.

skylerwshaw, 2024-09-18:
@fifieldt Can you help break down what your debug logs here demonstrate? I'm a software dev, though new to Meshtastic as of a week. Aiming to run some tests at various dBm levels between a t114 and a control Wisblock node. Would love insight into what logs are indicating that could benefit my analysis outside of just looking at whether the message was received by the control node. Thanks!

fifieldt, 2024-09-18:
Hi @skylerwshaw , the only real interesting messages are the last 5. 

We got a packet from the phone, plan to send it to lora channel 0, sent a 255 byte payload, then died after cleaning up and adding it to our list of sent packets. The half-printed line where it dies is from code that hasn't been changed in >6 months, so that's probably not the root of the problem.

lupusworax, 2024-09-18:
Got an Answer from Richard from Heltec directly after informing him about the ongoing problem: 

Hello.
First of all, the maximum transmit power of SX1262 should be 21±1.

Second, we noticed this phenomenon, which is more like a bug in the software, we found that sometimes your device shows failure to send, but in fact other nodes have received your message, this may be related to the forwarding mechanism of the software, this is our guess, we have just received a holiday, and have started testing this problem.

x81Reaper, 2024-09-19:
PROBLEM SOLVED !!! use the actual BETA app from the Play store !

x81Reaper, 2024-09-19:
Hello dear community, it is definitely not a hardware issue. I have fixed it on 3 devices. How? nrf erase multiple times, firmware 2.5, then back to 2.4.2 then back to 2.5, deleted again and back to 2.4.2 and now I can send the full number of characters :)

DTopp, 2024-09-19:
> Hello dear community, it is definitely not a hardware issue. I have fixed it on 3 devices. How? nrf erase multiple times, firmware 2.5, then back to 2.4.2 then back to 2.5, deleted again and back to 2.4.2 and now I can send the full number of characters :) Battery level over 90%

Couldn't get your method to work for me unfortunately although it did perform a little better than it has been.
[image]

lupusworax, 2024-09-19:
> Hello dear community, it is definitely not a hardware issue. I have fixed it on 3 devices. How? nrf erase multiple times, firmware 2.5, then back to 2.4.2 then back to 2.5, deleted again and back to 2.4.2 and now I can send the full number of characters :) Battery level over 90%

That sounds wild, there must be a better solution, that like rolling some dice and hoping to get a pair at some point??

s56oa, 2024-09-19:
> > Hello dear community, it is definitely not a hardware issue. I have fixed it on 3 devices. How? nrf erase multiple times, firmware 2.5, then back to 2.4.2 then back to 2.5, deleted again and back to 2.4.2 and now I can send the full number of characters :) Battery level over 90%
> 
> That sounds wild, there must be a better solution, thats like rolling some dice and hoping to get a pair at some point??

Sounds like some sort of timing issue?

My Heltec T114 has the same problem. 
Tested with full length (228 bytes) messages. Super and dandy up to 10 dBm. Above this it gets into trouble.

garthvh, 2024-09-19:
Sure seems like a hardware problem still.

lupusworax, 2024-09-20:
Update from Heltec: 

Lupus.
We are working on this issue and do not yet know the specific reasons. However, this can be temporarily improved in two ways:
1. Lower DBM, as you said, we wrote a firmware that limits DBM, and it has a much higher delivery success rate.
2. Changing the channel, the device seems to perform poorly on meshtastic's default band "Long-Fast", and appears to perform normally on the other channels so far.

Richard

richard@heltec.cn

x81Reaper, 2024-09-20:
but why could I then fix the error? If it was a hardware error, it would be wrong on all devices because it would then be a design error

garthvh, 2024-09-20:
> but why could I then fix the error? If it was a hardware error, it would be wrong on all devices because it would then be a design error

nobody can replicate your solution, it working at reduced power indicates this is hardware related.

x81Reaper, 2024-09-20:
If it were a hardware defect, several hundred devices would be affected and I think you would read more about it on the Internet (not just meshtastic). I have my solution to share, which worked for me.

efm-7, 2024-09-20:
Thanks @lupusworax 

PS. I realized the enable pin for that regulator is pulled down by default and controlled by the nRF, i'm sure the heltec folks can figure it out though. 

Another idea - Could perhaps an idle-state dump of the hardware config registers across other, known to be reliable, nRF devices such as the RAK4630, T-echo, etc..., be done?

lupusworax, 2024-09-20:
> If it were a hardware defect, several hundred devices would be affected and I think you would read more about it on the Internet (not just meshtastic). I have my solution to share, which worked for me.

How would you know your T114 is affected? 
I am pretty sure that all T114 are affected so far, but with meshtastic you never have a "perfect" feedback about message delivery, I would probably not have encountered this problem  for several months if i haven't read about it.  As I mostly dont write long messages  and if I write them I  would not have known if the failed delivery is caused by a hardware issue. ( I would have suspected pretty much everything else)  So as this issue just surfaced recently most people are not even aware of it.

efm-7, 2024-09-20:
> If it were a hardware defect, several hundred devices would be affected and I think you would read more about it on the Internet (not just meshtastic). I have my solution to share, which worked for me.

Personally, I don't doubt you, but it's not really a solution or fix unless the mechanisms are well-understood, otherwise it's just a "YMMV" remedy.

Your successful outcomes may rely on some sort of edge case that we don't know of, but are still valuable anecdotes! we need these clues :)

x81Reaper, 2024-09-20:
I had this problem! and then tried everything to fix it, as I described, I was previously unable to send more than a maximum of 45 characters, now I can send the full number of characters normally at full transmission power (22) if it was a hardware problem I couldn't fix it by simply deleting the software.

lupusworax, 2024-09-20:
> I had this problem! and then tried everything to fix it, as I described, I was previously unable to send more than a maximum of 45 characters, now I can send the full number of characters normally [...]

I am wondering, every single device I built and configured, for 868_EU the dBm setting was 27 (stock)  I never touched that value, Richard from heltec told me that  it should not be more then 21 +/- 1 just like you said yours is on 22.  How comes that 868_EU has a stock value of 27?

efm-7, 2024-09-20:
> I had this problem! and then tried everything to fix it, as I described, I was previously unable to send more than a maximum of 45 characters, now I can send the full number of characters normally [...]

Hence why I think it's worthwhile looking at what the config registers are doing... my hunch is that there may be something new/novel in the "offending" firmware that, once set, is out of reach in other versions of the firmware, and a firmware reload dance like you describe can sometimes get it to unstick? 🤷🏻‍♂️

mverch67, 2024-09-20:
> but why could I then fix the error? If it was a hardware error, it would be wrong on all devices because it would then be a design error

You are using LongModerate and not LongFast where the error occurs.

mverch67, 2024-09-20:
> I am wondering, every single device I built and configured, for 868_EU the dBm setting was 27 (stock) I never touched that value, Richard from heltec told me that it should not be more then 21 +/- 1 [...]

27 is the maximum value allowed for EU_868. When the power is configured to the hw driver it is limited to what the chipset is capable of, i.e. 22. But this value is not shown on the UI.

spicebagger, 2024-09-20:
> Hello dear community, it is definitely not a hardware issue. I have fixed it on 3 devices. How? nrf erase multiple times, firmware 2.5, then back to 2.4.2 then back to 2.5, deleted again and back to 2.4.2 and now I can send the full number of characters :) Battery level over 90%

I have tried this, and can confirm it works.  I can now send the full amount of characters on full power.

x81Reaper, 2024-09-20:
> > but why could I then fix the error? If it was a hardware error, it would be wrong on all devices because it would then be a design error
> 
> You are using LongModerate and not LongFast where the error occurs.

not only it works on every mode !

x81Reaper, 2024-09-21:
> Someone who has a tantalum capacitor (10+ uF) can try if it helps to put it across VDD_3V3 and GND. It can't do any harm, as long as you put it the right way round.
> 
> Maybe add `#define LORA_DIO1 SX126X_DIO1` if it is something to do with sleep interrupts.
> 
> Or Heltec flashed an old or custom version of the firmware, clear main flash then reupload latest version

Exactly what I mean 
Delete everything on this "Drive" 
and make an clean install

RichardHeltec, 2024-09-21:
> Hi Everyone, I was analyzing the schematic and and comparing it to the Wisblock etc...
> 
> I don't think there is enough isolation between the nRF and SX chips on the T114
> 
> One thing that could be tried is to pick off L4 inductor and on the and tie the nRF side to the auxiliary Ve_3.3v which has its own regulator
> 
> I am willing to risk my own T114 to give this a shot once i'm feeling brave enough to work at such small levels
> 
> ![image](https://private-user-images.githubusercontent.com/1829090/369427073-de953f1c-8032-4061-904d-4d575c4e9083.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY4ODczNDMsIm5iZiI6MTcyNjg4NzA0MywicGF0aCI6Ii8xODI5MDkwLzM2OTQyNzA3My1kZTk1M2YxYy04MDMyLTQwNjEtOTA0ZC00ZDU3NWM0ZTkwODMuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTIxVDAyNTA0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcxNGZjZTk3ZTIwNGUzNTFjZGNkODgxOGYxZDY4ODJmNzlhMTA5MTUyZGQ3ZDAzZTk1MTVhMjJkMjBmZDYyMDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.MyHSTSIz7J7GR7jdbl3J7YF-uTKWNeRWHEiqkbY6Ows)

Unfortunately, it doesn't seem to be working. Thanks for your help, we are also working on this over the weekend, and it seems that the band "250K/SF11" is causing mischief.

RichardHeltec, 2024-09-21:
Hello everyone.
We made a version of the firmware to alleviate this issue temporarily:
[Download link](https://resource.heltec.cn/download/Mesh_Node_T114/firmware/firmware-mesh-node-114-2.4.3-20240921.uf2)

We will continue to follow up on this situation and look forward to more feedback from you.
If you have any ideas that you would like to be verified, or need any help or consultation, you can contact us at: support@heltec.cn

s56oa, 2024-09-21:
To add, how I made it work using official mesthastic firmware from flasher.mesthastic.org:

1. Go to Official Flasher https://flasher.meshtastic.org/
2. Select Heltec Mesh Node T114
3. Near the green Flash button there is a trashcan icon, click on this thrashcan icon (erase Flash of Heltec Mesh Node T114)
4. Do the erasing of the flash by downloading Flash Erase UF2 file to the unit
5. After erasing is complete just download and flash latest official firmware (I used official latest beta firmware-mesh-node-114-2.4.3-20240921.uf2)

Now my Heltec T-114 was able to send 227 bytes of text with default 27 dBm power setting using Long Range - Fast preset for EU 868 MHz.

 
[image]

Haven't tried the RichardHeltec Download link file.

@RichardHeltec  Maybe if you can explain to the dev team what was changed in your firmware so it can be pushed to official release.

x81Reaper, 2024-09-21:
exactly what I have said...delete everything.and flash new

lupusworax, 2024-09-21:
I tried this with 2.5.0.9 Alpha and it did not work for me.

lupusworax, 2024-09-21:
> Maybe if you can explain to the dev team what was changed in your firmware so it can be pushed to official release.

I think they lowered the dBm to 11.

aljaxus, 2024-09-21:
> To add, how I made it work using official mesthastic firmware from flasher.mesthastic.org:
> 
>     1. Went to Official Flasher https://flasher.meshtastic.org/
> 
>     2. Selected Heltec Mesh Node T114
> 
>     3. Near the green Flash button there is a trashcan icon, I click on this trashcan icon (erase Flash of Heltec Mesh Node T114)
> 
>     4. I did the erasing of the flash by downloading Flash Erase UF2 file to the unit
> 
>     5. After erasing was completed I just download and flashed latest official beta firmware (I used official latest beta firmware-heltec-mesh-node-t114-2.4.2.5b45303.uf2 file)
> 
> Now my Heltec T-114 is able to send 227 bytes of text with default 27 dBm power setting using Long Range - Fast preset for EU 868 MHz.

Can confirm this worked for me too on multiple nodes.

garthvh, 2024-09-22:
> Hello everyone.
>
> We made a version of the firmware to alleviate this issue temporarily:
>
> [Download [...]

Where is the code for this fork located at?

DTopp, 2024-09-22:
> exactly what I have said...delete everything.and flash new

Gave it another try and it still didn't work for me. All seems a bit weird.

DTopp, 2024-09-22:
> Hello everyone. We made a version of the firmware to alleviate this issue temporarily: [Download link](https://resource.heltec.cn/download/Mesh_Node_T114/firmware/firmware-mesh-node-114-2.4.3-20240921.uf2)
> 
> We will continue to follow up on this situation and look forward to more feedback from you. If you have any ideas that you would like to be verified, or need any help or consultation, you can contact us at: [support@heltec.cn](mailto:support@heltec.cn)

This firmware does work although I'm a little suspicious as to why it works. Is it true that it's just the dBm that has been lowered and what effect will that have on range if so?

lupusworax, 2024-09-22:
I am confused in general when it comes to the NRF52 chipset, its my first of its kind  and the whole "flashing" stuff is causing me headaches. 
When I was trying the posted solution by deleting a few times, flashing older firmware  back and forth, erasing again, on my device the possibility to turn gps on and off with 3 button presses stopped working,

 no matter what I did, while double pressing sending adhoc worked, gps was not reactive to 3 button presses.  I then went on wild erasing, flashing one after another firmware back and forth, erasing again 4 times at once as it suddenly at some point started working again......... This is wild.  
With the ESP32 devices I never had anything similar, when I flashed a new firmware I was assured that all the old hardware was gone and no issues remained........but with the NRF52  ( or maybe just with the heltec t114) this is just a hot mess as it seems it never really overwrites the old firmware reliable, is this also the same with RAK based nodes??

lupusworax, 2024-09-22:
> This firmware does work although I'm a little suspicious as to why it works. Is it true that it's just the dBm that has been lowered and what effect will that have on range if so?

The info about lowering dBm to something around 11 was the last info I got from heltec before I told them it would be best if they directly post their findings on here other then writing messages to me ( I was the one contacting heltec about the problem in the first place)

Well if the max possible output is 21 +/- 1 you can be sure that its a lot less with 11 then it would be capable off and so range is therefor limited of course.  
When you check with an SDR you can see that it comes in a lot quieter then with max power.

s56oa, 2024-09-22:
@lupusworax I had issues with some ESP32 devices before erasing fash, too. So this is nothing new. 

My T114 was shipped with some non-Meshtastic firmware from the factory and this might be the original offender.
Erasing the flash seems to clean and tidy up the "house" before the new furniture (official Meshtastic firmware) is brought in.

garthvh, 2024-09-22:
@RichardHeltec where is the code?

s56oa, 2024-09-22:
Having the code change would be beneficial.
But as evidenced in the field, that erasing the flash makes even the official Meshtastic firmware work, it would be of interest to find the offending culprit. It seems it can be found in the contents of flash memory from the factory.

[140 more replies]

closed 2024-09-29 by garthvh