meshtastic/firmware #4725: [Bug]: Buffer too small publishing enqueued MQTT messages

jrouvier, 2024-09-15:
### Category

Other

### Hardware

Station G2

### Firmware Version

2.5.1.88935296

### Description

When initially connecting to MQTT a queued message for MQTT that is rather long will cause a panic and reboot.  In [src/mqtt/MQTT.cpp:471](https://github.com/meshtastic/firmware/blob/master/src/mqtt/MQTT.cpp#L471) the buffer is defined as:
```C
        static uint8_t bytes[meshtastic_MeshPacket_size + 64];
        size_t numBytes = pb_encode_to_bytes(bytes, sizeof(bytes), &meshtastic_ServiceEnvelope_msg, env);
```
`meshtastic_MeshPacket_size` is 367, which gives a sum of 431.  This probably should be changed to:
```C
        static uint8_t bytes[meshtastic_MqttClientProxyMessage_size];
        size_t numBytes = pb_encode_to_bytes(bytes, sizeof(bytes), &meshtastic_ServiceEnvelope_msg, env);
```
(`meshtastic_MqttClientProxyMessage_size` is 501)

Note that other bugs complaining about "reboot loops after enabling wifi" or similar may also be due to this bug.

### Relevant log output

```Shell
DEBUG | ??:??:?? 49 [Router] MQTT onSend - Publishing encrypted message
INFO  | ??:??:?? 49 [Router] MQTT not connected, queueing packet
INFO  | ??:??:?? 53 [mqtt] Using non-TLS-encrypted session
INFO  | ??:??:?? 53 [mqtt] Attempting to connect directly to MQTT server mqtt.bayme.sh, port: 1883, username: meshdev, password: large4cats
INFO  | ??:??:?? 53 [mqtt] MQTT connected
INFO  | ??:??:?? 53 [mqtt] Subscribing to msh/US/bayarea/2/e/Rouvier/+
INFO  | ??:??:?? 53 [mqtt] Subscribing to msh/US/bayarea/2/e/PKI/+
DEBUG | ??:??:?? 53 [mqtt] Publishing enqueued MQTT message
ERROR | ??:??:?? 53 [mqtt] Panic: can't encode protobuf reason='bytes size exceeded'
��@INFO  | ??:??:?? 1 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 1 Booted, wake cause 0 (boot count 1), reset_reason=reset
```

thebentern, 2024-09-15:
Great find! I'll push up a PR fixing the buffer allocation.

thebentern, 2024-09-15:
After taking a look at this again, it appears we might be overflowing MqttClientProxyMessage_data underneath this. I'll poke at it.

closed 2024-09-15 by thebentern

chairmanfaust, 2024-12-12:
I hate to resurrect a dead topic but I'm finding that my station g2 when configured to use wifi to connect to my mqtt server still drops the wifi after a few days and I'm forced to power cycle it

fifieldt, 2024-12-13:
Probably best to raise a new issue so we don't lose your report :)

chairmanfaust, 2024-12-13:
Thanks I will... I have two station g2s and I was just about to put this on my roof when the wifi kicked out

PR #4726 (2024-09-15): Expand buffer to MqttClientProxyMessage_size
While not technically an MqttClientProxyMessage, expanding the buffer to this size should give us a more room for ServiceEnvelopes with large MeshPackets and long topic names + gateway_ids. We may need to place some long term limitations on size of topic names, because these can get close to the MTU for bluetooth packets. So we'll need to circle back, but this should work for now hopefully.

Closes #4725