meshtastic/firmware #4730: [Bug]:  Heltec T114 turning on/off GPS and  sending ad hoc signal not working

lupusworax, 2024-09-16:
### Category

Other

### Hardware

Other

### Firmware Version

2.5

### Description

Pressing the Program  button 3 times should deactivate GPS, its not working same with sending out an ad hoc signal by pressing it 2 times.

jumpgate, 2024-09-16:
Check the screen that shows your location info once GPS gets a lock. I see "GPS is disabled" after pressing the button 3 times, and doing it again shows the location info. A 2 button press does send the position announcement based on what I see in the log and on another node (last heard time will update). The notifications for each operation just don't show up where you'd expect them to be. This is with the latest 2.5.0 release.

lupusworax, 2024-09-16:
Will update and check again, would it be possible for the adhoc info to be shown. I dont always have a second node at hand to check if the double button press was done correctly, my experience is that more often the system is kind of busy in the background and does not recognize both presses or even disabled/ enabled GPS. with it being posted on the screen like it was would be really helpful.  Havin to check the log file every time to see if it was updated  is kind of a downgrade from what I am used to.

lupusworax, 2024-09-16:
Alright, your correct it works now, I can swear I updated 2 version of 2.5 and it did not work but this now works ;)
I also tested the adhoc update, it does update checking logs and on another node but the info about sending a positive ad hoc update on the display still would be great!

closed 2024-09-17 by fifieldt

todd-herbert, 2024-09-17:
I think it's just not as obvious right now that these double or triple presses are actually talking place, because of the confirmed issue with the log buffer: https://github.com/meshtastic/firmware/issues/4731