meshtastic/firmware #4731: [Bug]:  Heltec T114 not showing Logs of last joined nodes or events

lupusworax, 2024-09-16:
### Category

Other

### Hardware

Other

### Firmware Version

2.5

### Description

Overview page showing battery symbol in the left top corner and  channel ( longfast) below, GPS Status and Device ID in tht top right used to show recent events like turning gps on and off, sending ad hoc signal, and recent joined/seen nodes. 
They are completely gone and dont show up at all. 
[image]

todd-herbert, 2024-09-17:
I can replicate the issue here. I'll look into it.

fifieldt, 2025-02-08:
Confirming on T114 screen you don't see the "Joined [nodename]" messages on that page. Latest master.

todd-herbert, 2025-02-09:
Some background info on this one: I remember that this "log buffer" feature was provided by the OLED library, but something changed upstream meaning that we couldn't use it anymore. I think I quickly wedged the old feature back into our fork of the library, but it would probably make more sense to reimplement it entirely, at the Meshtastic fw layer instead.

Xaositek, 2025-06-24:
With the release of the technical preview of Meshtastic 2.7, this feature has been removed at this time from all devices in favor of the new Node List.

closed 2025-06-24 by Xaositek