meshtastic/firmware #4737: [Bug]: build dependency adafruit-nrfuitil.exe flagged by AV software

powersjcb, 2024-09-16:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

main

### Description

adafruit-nrfuitil.exe flagged by AV software

Blocks firmware dev work on any machine with locked down AV.

https://www.virustotal.com/gui/file/9637a4d099f4a99cbbdfb14142345776582dfbd11d4f357748055b75006ac8c6/details

Related issue https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/424 The fix seems to be using a different version of the library. TBD if that actually fixes the root cause or if the virus scan definitions just don't have the checksums for the latest master builds. :-\

[image]

powersjcb, 2024-09-16:
It generally sounds like this is an issue bc virus profiles commonly bundle python code into an .exe (as this dependency has)

Almost certainly a false positive, but it would still be nice to figure out a way to resolve this.

garthvh, 2024-09-16:
What antivirus triggered the false positive? They all have a way to submit software, I have done windows defender for the old desktop flasher.

powersjcb, 2024-09-16:
Multiple vendors are flagging it as malware per VirusTotal
* Sentinel One
* [Bkav Pro](https://www.bkav.com/bkav-pro)
* Skyhigh (SWG) (Zillya) [form to report false positives](https://www.avira.com/en/analysis/submit)

closed 2024-11-24 by github-actions[bot]