meshtastic/firmware #4738: Addtional Sensor for Farm Animals Health

nextechDI, 2024-09-16:
Greetings, congratulations to all the developers and collaborators, this project is fantastic, I wanted to contribute about use cases and sensors, I think that the meshtastic firmware for monitoring farm animals has a lot of potential by allowing several sensors to be integrated with the same micro, but I would recommend adding the MLX90614 temperature sensor to the telemetry firmware, which is the most suitable for sensing temperature in animals. I would also recommend adding the MAX30102 heart rate sensor and oximeter, which offers an excellent quality-price ratio for this function. These two sensors together Those that already include the meshtastic firmware would allow it to perform very well as a farm animal tracking device.

dhskinner, 2024-10-02:
Hi @nextechDI I have exactly zero of those sensors but don't mind at all contributing an integration if you're ok to do some testing. From a quick skim both have reasonable library support:
- [Adafruit-MLX90614-Library](https://registry.platformio.org/libraries/adafruit/Adafruit%20MLX90614%20Library)
- [sparkfun/SparkFun MAX3010x Pulse and Proximity Sensor Library](https://registry.platformio.org/libraries/sparkfun/SparkFun%20MAX3010x%20Pulse%20and%20Proximity%20Sensor%20Library)

Also consider that you can get other info from the accelerometer such as what-do-cows-do-when-no-one-is-looking?  (if only we had the data... those cows are up to something I'm sure)

@thebentern might need to add some new fields to [https://github.com/meshtastic/protobufs](https://github.com/meshtastic/protobufs) for oxygen level and pulse rate? Appreciate your thoughts

thebentern, 2024-10-02:
@dhskinner that was super spooky timing 👀 
https://github.com/meshtastic/protobufs/pull/590

fifieldt, 2024-10-02:
Popular issue :) I've also just acquired these sensors so can contribute to code and testing.

thebentern, 2024-10-02:
I'll push a branch after the protobuf classes regenerate and we'll collab.

fifieldt, 2024-10-07:
Hi @nextechDI , we've just about got a prototype for telemetry using those sensors you mentioned. Do you have devices available to help test in the real world? We can send updated firmware...

nextechDI, 2024-10-07:
> Hi @nextechDI I have exactly zero of those sensors but don't mind at all contributing an integration if you're ok to do some testing. From a quick skim both have reasonable library support:
>
>     * [...]

Greetings, thank you for your contribution, I bought the sensors, they should arrive in 2 or 3 weeks, I can also help with the tests, yes I agree about the accelerometer, the two best options are the LIS3DH and the LSM6DS3 which also has a gyroscope , since movement and angle help to identify the type of head movement among ordinary people, complaints of discomfort, drinking water, agitation due to an attempted robbery hahahaha, of course all of this is evaluated based on the data from the other sensors, about "@thebentern might need to add some new fields to https://github.com/meshtastic/protobufs for oxygen level and pulse rate? Appreciate your thoughts" at least two fields must exist to save the oxygen concentration records ("Calculated SPO2") and for pulse rate ("Calculated heart-rate") but recently I am getting familiar with meshtastic's working methods. Thank you for the support of the project, after this project I plan to build an economical clone hardware for meshtastic aimed at low energy consumption and cost for this project and other similar ones, I already have the chips selected, greetings.

nextechDI, 2024-10-07:
> Hi @nextechDI , we've just about got a prototype for telemetry using those sensors you mentioned. Do you have devices available to help test in the real world? We can send updated firmware...

Greetings, thank you for your initiative, I plan to use the seeed wio tracker 1110, along with the MAX30102 and MLX90614 sensors, the wio tracker 1110 already includes an accelerometer to do the tests, I already bought them, they should arrive in 2 or 3 weeks to arrive, I plan to do Field tests with animals, when you receive notice, and I also buy the results of the tests, if you have the firmware for the Wio Tracker 1110 with support for those excellent sensors. I plan to build a Wio Tracker 1110 clone card with equivalent, cheaper chips. thanks for helping on this project.

closed 2024-10-08 by thebentern