meshtastic/firmware #4752: [Feature Request]: ST7789 TFT Rotate 90 Degrees

el-samiyel, 2024-09-17:
### Platform

ESP32

### Description

Hi,

I have a custom ESP32 ST7789 combo. I want to be able to rotate the 240x40 display by 90degrees. Is this possible?

Tried various software build configurations, but no luck.

todd-herbert, 2024-09-18:
Just for some background info, what does your platformio.ini and variant.h file look like so far?

Szetya, 2024-09-18:
In the variant.h file you can specify the display orientation with the TFT_OFFSET_ROTATION macro. 
Its value can be 0-7. Between 0-3 rotated 0, 90, 180, 270 degrees, between 4-7 rotated and mirrored.

el-samiyel, 2024-09-18:
> In the variant.h file you can specify the display orientation with the TFT_OFFSET_ROTATION macro. Its value can be 0-7. Between 0-3 rotated 0, 90, 180, 270 degrees, between 4-7 rotated and mirrored.

Hi thanks for the detail, I did think this, but the changes don't take, the landscape display remains persistent. Here is a code snip, let me know if you see anything wrong:

`#define USE_ST7789
#define TFT_HEIGHT 240
#define TFT_WIDTH 240
#define TFT_OFFSET_X 0
#define TFT_OFFSET_Y 0
#define TFT_OFFSET_ROTATION 1`

Just tried with my custom NRF52 and get the same results.

Szetya, 2024-09-18:
As Todd-Herbert also asked, he can you copy the full platformio.ini and variant.h file?

Szetya, 2024-09-18:
> Just for some background info, what does your platformio.ini and variant.h file look like so far?

I can think that -DT_WATCH_S3 is in the platformio.ini file and the directive at the bottom of TFTDisplay.cpp overrides the orientation.

todd-herbert, 2024-09-18:
If you define `USE_ST7789`, you will use the ST7789 library contributed recently by Heltec, specifically for their T114 and T190 devices. Try removing this line. 
Because Heltec created that library specifically for their two devices, it isn't very convenient to configure it right now.
(Maybe we could make it more convenient in future?)

It seems like most devices with ST7789 are using the LovyanGFX library. I'm not personally familiar with it, but it seems like it should be used if you follow [variant.h](https://github.com/meshtastic/firmware/blob/master/variants/picomputer-s3/variant.h#L29) file for the picomputer-s3

I'm not sure if LovyanGFX is compatible with NRF52 or not. If you *do* need to Heltec's ST7789 library, I can give a general description of how to rotate the display, but the process is quite complicated.

el-samiyel, 2024-09-18:
Oh ok, understood I'm starting to understand how things works now in terms of how the libs are used (but not enough to solve this).

With that line commented, the screen doesn't turn on anymore, no backlight and as far as I can see (with a torch) nothing displayed).

`//#define USE_ST7789
#define ST7789_NSS 11
#define ST7789_RS 12  // DC
#define ST7789_SDA 41 // MOSI
#define ST7789_SCK 40
#define ST7789_RESET 2
#define ST7789_MISO -1
#define ST7789_BUSY -1
#define VTFT_CTRL 3
#define VTFT_LEDA 15
#define TFT_BACKLIGHT_ON LOW
#define ST7789_SPI_HOST SPI1_HOST
#define SPI_FREQUENCY 40000000
#define SPI_READ_FREQUENCY 16000000
#define TFT_HEIGHT 240
#define TFT_WIDTH 240
#define TFT_OFFSET_X 0
#define TFT_OFFSET_Y 0
#define TFT_OFFSET_ROTATION 1`

el-samiyel, 2024-09-18:
> If you define `USE_ST7789`, you will use the ST7789 library contributed recently by Heltec, specifically for their T114 and T190 devices. Try removing this line. Because Heltec created that library specifically for their two devices, it isn't very convenient to configure it right now. (Maybe we could make it more convenient in future?)
> 
> It seems like most devices with ST7789 are using the LovyanGFX library. I'm not personally familiar with it, but it seems like it should be possible to use it instead of the Heltec library if you follow [variant.h](https://github.com/meshtastic/firmware/blob/master/variants/picomputer-s3/variant.h#L29) file for the picomputer-s3
> 
> I'm not sure if LovyanGFX is compatible with NRF52 or not. If you _do_ need to Heltec's ST7789 library, I can give a general description of how to rotate the display, but the process is quite complicated.

Understood, I'm starting to understand how things works now in terms of how the libs are used (but not enough to solve this issue).

With that line commented, the screen doesn't turn on anymore, no backlight and as far as I can see (with a torch) nothing displayed).

`//#define USE_ST7789 #define ST7789_NSS 11 #define ST7789_RS 12  // DC #define ST7789_SDA 41 // MOSI #define ST7789_SCK 40 #define ST7789_RESET 2 #define ST7789_MISO -1 #define ST7789_BUSY -1 #define VTFT_CTRL 3 #define VTFT_LEDA 15 #define TFT_BACKLIGHT_ON LOW #define ST7789_SPI_HOST SPI1_HOST #define SPI_FREQUENCY 40000000 #define SPI_READ_FREQUENCY 16000000 #define TFT_HEIGHT 240 #define TFT_WIDTH 240 #define TFT_OFFSET_X 0 #define TFT_OFFSET_Y 0 #define TFT_OFFSET_ROTATION 1`

So I think I need to use Heltec's ST7789, yes please, any direction will be much appreciated.

todd-herbert, 2024-09-18:
>Understood, I'm starting to understand how things works now in terms of how the libs are used (but not enough to solve this issue).

It still confuses me myself! 

Just one other option to explore: other than the Heltec T114, I can see *one* other NRF52 device with a TFT display: [RAK10701](https://github.com/meshtastic/firmware/tree/master/variants/rak10701).
It uses yet another different library (bodmer/TFT_eSPI).

___

As for Heltec's ST7789 library, it seems like the display config is hard-coded into the library.
You can make temporary changes to this by editing the `.pio/libdeps/<your hardware>/ESP8266 and ESP32 TFT driver for st7789 displays/src/ST7789Spi.h`

Looking through this file, you will see several occurrences of `ST77XX_MADCTL_MV`. This swaps the rows and columns, effectively rotating the display by 90 degrees. Removing references to `ST77XX_MADCTL_MV` should rotate the display.

It's quite likely that removing `ST77XX_MADCTL_MV` will mirror the display, or that the display will not rotate in the direction you hope. You may need to experiment with adding or removing `ST77XX_MADCTL_MX` and `ST77XX_MADCTL_MY` to correct this.

You may also need to adjust the display resolution. You would need to change the values being sent with `ST77XX_CASET` and `ST77XX_RASET` in `sendInitCommands()`.

I'm attaching the datasheet for ST7789, in case it helps to see a more technical explanation of these config options.
[ST7789.pdf](https://github.com/user-attachments/files/17041997/ST7789.pdf)

Note that changes made to your `.pio` folder like this are not permanent. They will likely be overwritten at some point. In order to preserve the changes, you might want to move the ST7789Spi.h file to the `src` folder, and use this local copy by removing the library from the lib_deps section of your platformio.ini file.