meshtastic/firmware #4758: [Bug]: Heltec Wireless Tracker v1.1 Serial non-responsive after some time (sleep?)

65a, 2024-09-18:
### Category

Serial

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.5.0.9ac0e26

### Description

Serial works right after boot. If I leave it unattended for some time, and then try to connect serial, the serial terminal hangs and nothing is printed. It seems like the peripheral may be in a bad state after sleep or something.

### Relevant log output

```Shell
n/a (that's the problem)
```

fifieldt, 2024-09-18:
testing.
* 30mins : OK

65a, 2024-09-18:
An easier way to reproduce here with two recent boards, is to let them boot and attach to /dev/ttyACM later. They seem to stay working if you attached serial "early" enough.

fifieldt, 2024-09-18:
ah, good to know.

fifieldt, 2024-09-19:
Hmm. I left one overnight and it came up OK. Trying again.

65a, 2024-09-19:
Interesting, thanks for trying. I will try a different computer/terminal emulator/cable, worried there's a USB power saving issue, or something else weird going on, but I've used the same machine extensively for embedded dev (e.g. t-beam works fine)

fifieldt, 2024-09-19:
No problems - intermittent issues are hard!

Just put on the latest master and ran for an hour and it was still fine.

caveman99, 2024-09-19:
> Interesting, thanks for trying. I will try a different computer/terminal emulator/cable, worried there's a USB power saving issue, or something else weird going on, but I've used the same machine extensively for embedded dev (e.g. t-beam works fine)

Difference could be T-BEam uses a hardware UART/USB Chip, while the Wireless Tracker uses a software UART from the ESP32-S3 CPU.

65a, 2024-09-21:
@fifieldt I flashed HEAD and I can't seem to reproduce this anymore locally, so that seems good. `2.5.0.9ac0e26` is still the alpha on flasher.meshtastic.org, with the serial issues it was quite hard to flash anything else, 1200bps reset was absolutely required before anything, including esptool. So if anyone else encounters this, flash a different version after doing 1200bps reset via the web flasher.

closed 2024-09-21 by 65a