meshtastic/firmware #4760: [Feature Request]: Dual Frequency Support 900MHz + 2.4GHz

b8b8, 2024-09-18:
### Platform

NRF52, ESP32, Cross-Platform

### Description

This is just an idea I wanted to get in writing; i dont expect it to be worked on anytime soon.

My idea is to slave a 2.4GHz LoRa radio to our standard NRF or ESP32 boards. 
The main purpose will be to drastically increase the bandwidth available for city and high congestion events while falling back to standard 900MHz when appropriate. 

The two radios show up as a single combined radio (thus the master/slave relationship) to the app with two default public channels. 
The standard LongFast 900 LoRa channel and the new 2.4GHz LoRa channel would show up on the message list. 

The major change is now all direct messages (and maybe private-primary channel telemetry) would first be tried at the much higher bandwidth 2.4 channel and if no acknowledgement is received, then it will resend the DM on the 900 LoRa channel. The two frequencies would not actually be bridged directly so you dont have to worry about dumping a massive amount of 2.4 traffic onto the more fragile 900 band. Since both public channels would still exist people can still chat on both to the public independently. 

Without breaking backward compatibility and still allowing users to interact with nodes that dont have 2.4 radios, this could dramatically increase available bandwidth in congested areas (and offload private telemetry with public fallback). No changes would need to be made say when hiking and the 2.4 band cant make contact, as it would fall back to the longer range 900 band.

caveman99, 2024-09-19:
Parts of this could be implemented with a LR1121 based wideband radio. I don't think it can listen to 900 and 2.4 at the same time though. I have a Lilygo T3 S3 LR1121 incoming and after initial support we can revisit this.

b8b8, 2024-09-19:
Excellent. This also provides a little bit more worldwide compatibility by default as I believe 2.4 LoRa is legal in all countries like wifi is.

b8b8, 2024-11-24:
Please pin this feature for future consideration.

xznhj8129, 2025-03-05:
This is a very interesting option especially if we look into adapting GemX ELRS hardware, two LR1121 modules integrated into a single unit gives the opportunity to operate on both bands at the same time