meshtastic/firmware #4768: [Board]: M5Stack CoreS3 + LoRa868_V1.1 Module

lboue, 2024-09-18:
### SOC

ESP32

### Lora IC

SX1276

### Product Link

https://shop.m5stack.com/products/lora-module-868mhz-v1-1

### Description

Setup
- [M5Stack CoreS3](https://shop.m5stack.com/products/m5stack-cores3-esp32s3-lotdevelopment-kit)
- [LoRa868_V1.1 Module](https://shop.m5stack.com/products/lora-module-868mhz-v1-1)

Coopersmith-24601, 2024-09-19:
It's already being worked on.  I personally can't hardly wait to see it.  It's going to be an incredibly interesting opportunity to add a bajillion different types of sensors.

lboue, 2024-09-19:
Do you know who's working on it? I was thinking of starting too.

Coopersmith-24601, 2024-09-19:
Yes, NM.Patriot over on the Discord.

https://discord.com/channels/867578229534359593/931765413350613052/1285356293485494375

lboue, 2024-09-22:
M5Stack CoreS3 has AXP2101 power management like [tbeam-s3-core](https://github.com/meshtastic/firmware/blob/master/variants/tbeam-s3-core/variant.h) variant.

lboue, 2024-09-22:
### AXP2101 usage
|           |M5Stack<BR>CoreS3  |           |
|:---------:|:--------------------:|:-----------------:|
| ALDO1     |VDD 1v8            | ALDO1     |
| ALDO2     |VDDA 3v3           | ALDO2     |
| ALDO3     |CAM 3v3            | ALDO3     |
| ALDO4     |TF 3v3             | ALDO4     |
| BLDO1     |AVDD               | BLDO1     |
| BLDO2     |DVDD               | BLDO2     |
| DLDO1/DC1 |LCD BL             | DLDO1/DC1 |
| DLDO2/DC2 | ---               | DLDO2/DC2 |
| BACKUP    |RTC BAT            | BACKUP    |

closed 2024-11-30 by github-actions[bot]

Coopersmith-24601, 2024-12-10:
So, that's it then.  It just dies?

Sadness.