meshtastic/firmware #4775: [Bug]: Unclear Issues with LR1110; "my T-1000-E is deaf"

fifieldt, 2024-09-19:
### Category

Hardware Compatibility

### Hardware

Seeed Card Tracker T1000-E

### Firmware Version

2.5.0

### Description

As discussed in Discord over the past 3 weeks, there are multiple instances of people reporting issues with LR1110 radios. This issue is being created to serve as a central place for recording this information and coordinate testing.

caveman99, 2024-09-19:
> @Talie5in has identified #4621 as the breakpoint.

Are the other devices on the same firmware level or older/newer? We have radiolib changes scattered in there somewhere.

fifieldt, 2024-09-19:
Currently trying: 33eb073535c2bece2353b9fd76d0dda5d120ac05 with a later version of RadioLib...

fifieldt, 2024-09-19:
OK, https://github.com/meshtastic/firmware/commit/33eb073535c2bece2353b9fd76d0dda5d120ac05 with RadioLib 7.0.0 results in the bad outcome.

By my logic, using the good meshtastic commit with just radiolib changes means the issue might be in RadioLib!

fifieldt, 2024-09-19:
RadioLib 6.6.0 (working version) is May 29 https://github.com/jgromes/RadioLib/commit/954477b0c0589e6373692ce569ca117191667c30 .

Stepping forward through RadioLib with known-good Meshtastic version.

fifieldt, 2024-09-19:
June 1    7a0b5bae9906905f32cd840ec79b2665f509f717 - OK

Talie5in, 2024-09-19:
> > @Talie5in has identified #4621 as the breakpoint.
> 
> Are the other devices on the same firmware level or older/newer? We have radiolib changes scattered in there somewhere.

The one example is my WisMeshPockt (Rak4631) on 2.5.0 Alpha (through the entire time I was testing different 2.5.0 TechPreview build - it was out in my car (50m away) and was the only device it would hear past https://github.com/meshtastic/firmware/commit/33eb073535c2bece2353b9fd76d0dda5d120ac05

GUVWAF, 2024-09-19:
That RadioLib commit introduces differences in the IRQ flags handling in `startReceive()`, so if using Meshtastic commit https://github.com/meshtastic/firmware/commit/33eb073535c2bece2353b9fd76d0dda5d120ac05 it is expected to break, but PR #4621 also should handle those API changes.

fifieldt, 2024-09-20:
I amended the meshtastic code appropriate to deal with the IRQ flag changes.

fifieldt, 2024-09-23:
Potentially related: https://github.com/meshtastic/firmware/pull/4827

valzzu, 2024-09-25:
works fine on long fast but doesent work on long moderate.
wio tracker is the only node that can communicate with it.

GUVWAF, 2024-09-26:
The incompatibility between SX127x and LR1110 really seems to be because of our non-standard sync word. If I change it to the default private network (0x12), I can send and receive between them.

GUVWAF, 2024-09-26:
Confirmed `LONG_MODERATE` doesn't work between LR1110 and other radios, and it's likely due to low data rate optimization (LDRO), which is automatically enabled for SX126x and SX127x when the symbol length is longer than 16ms, but it's not yet there for LR11x0. I'll see if I can add it.

GUVWAF, 2024-09-26:
@valzzu `LONG_MODERATE` should work after we update RadioLib to include https://github.com/jgromes/RadioLib/pull/1237.

valzzu, 2024-09-26:
Awesome

GUVWAF, 2024-09-27:
@Talie5in @fifieldt Can you try if #4885 fixes the bad receive performance for you?

> In my testing, I amended the meshtastic code appropriately to deal with the IRQ flag changes.

Were you using `lora.startReceive(RADIOLIB_LR11X0_RX_TIMEOUT_INF, RADIOLIB_IRQ_RX_DEFAULT_FLAGS, RADIOLIB_IRQ_RX_DEFAULT_MASK, 0)`? If so, that only works from RadioLib commit https://github.com/jgromes/RadioLib/commit/3115fc2d6700a9aee05888791ac930a910f2628f (which is the one PR #4621 uses and is three commits after the commit you used).

Talie5in, 2024-09-28:
> Can you try if #4885 fixes the bad receive performance for you?

@GUVWAF Built against [699d81c](https://github.com/meshtastic/firmware/pull/4885/commits/699d81c2a14c429618f4541ac64ff1f284b23c22) in [PR#4885](https://github.com/meshtastic/firmware/pull/4885) and I am getting nodes appearing after a nodeDB reset.

Some logs attached - of note one interesting bits i've not seen before but are likely normal...

```WARN | 04:04:10 490 [Router] Alloc an err=5,to=0x850354e6,idFrom=0x1b7f84e4,id=0x25516740```

[meshtastic-log-2024-09-28T04-07-35.735Z.log](https://github.com/user-attachments/files/17172209/meshtastic-log-2024-09-28T04-07-35.735Z.log)

GUVWAF, 2024-09-28:
Great! 

`WARN | 04:04:10 490 [Router] Alloc an err=5` means "Max. retransmission reached".

fifieldt, 2024-10-01:
Many thanks @GUVWAF , coming back online a d should be able to test in the next few days.

fifieldt, 2024-10-13:
Forgot to comment - the LR1110 now works >10km after those changes. The remaining issue is indeed the SX1276 one.

egzumer, 2024-12-10:
I found an interesting article about syncword setup between sx127x and sx126x. May help with this issue.
https://blog.classycode.com/lora-sync-word-compatibility-between-sx127x-and-sx126x-460324d1787a

fifieldt, 2024-12-14:
Word on discord is we got a new contact with Semtech and are discussing this.

wojo, 2025-02-03:
Any update on the new contact from Semtech on a potential field deployable fix?

Tyler-2, 2025-02-21:
Conversation elsewhere implies we should just throw away our older devices. Kinda surprised at how dismissive a large swat of people are of the waste and expense of ditching boards, possibly batteries, and enclosures. Naturally my whole kit before the T1000 was Lora32's and old TBeams. It worked. I have a functioning mesh. And now I've been "Microsoft'd" by a new device that isnt compatible with a standard?

What's the general expectation from the smart people?

- Is the problem with T1000's or the older Semtechs being out of spec?
- Is it likely that the nature of the problem would allow either of them to be fixed via software?

garthvh, 2025-02-22:
The 1276 radio in question is 13 years old and does not perform nearly as well as the newer radio in a T-1000.  I get much better distance and far fewer crc errors from a T-1000 than those older devices.

Tyler-2, 2025-02-22:
> The 1276 radio in question is 13 years old and does not perform nearly as well as the newer radio in a T-1000. I get much better distance and far fewer crc errors from a T-1000 than those older devices.

That's great, but the TBeams and Lora32's that I, and many others, have in service are not 13 years old. I understand that newer devices are (hopefully) better, more capable, and more energy efficient - but they're also not free in any sense. There's nothing wrong with old devices.

Are you implying that the issue is with the SX1276's, and that supporting them is holding back some new evolution of LORA/Meshtastic? If this is the case of like, 802.11B not being supported on modern hardware anymore, fine, that makes sense. That's what I'm hoping to understand. But just because T1000's are "better" when talking amongst themselves is not a reason to accept the deprecation of older, in-spec, perfectly capable hardware. If there's some great new feature that the old radios don't support, which Meshtastic wants to move on to supporting, that could be a valid reason to abandon old hardware support, but that doesn't sound like what's happening here.

It sounds like this is a DEFECT of the T1000's, and I don't understand why peoples' responses to that are "But it's otherwise a better radio, you should throw away your old ones." The old ones worked fine until the T1000's came along. What changed? _Someone_ is out of spec, I presume.

> Currently, LR1110 radios are unable to receive Meshtastic packets from the older SX127x radios. Semtech is aware of this issue, and we're hopeful they'll provide a solution soon.

So, not looking for a review on the T1000's. I have two, and currently wish I didn't. Hoping someone can share their opinions on:

- Is the problem with T1000's or the older Semtechs being out of spec?
- Is it likely that the nature of the problem would allow either of them to be fixed via software?

garthvh, 2025-02-22:
Semtech recommended using a 1262 or LR1110 radio to improve performs and reduce errors.

Tyler-2, 2025-02-22:
Sounds like "Semtech is aware of this issue, and we're hopeful they'll provide a solution soon." is no longer true then, and their solution is "buy more of our products!"

garthvh, 2025-02-22:
> Sounds like "Semtech is aware of this issue, and we're hopeful they'll provide a solution soon." is no longer true then, and their solution is "buy more of our products!"

Your T-1000 with a pcb antenna performs much better than those older devices with larger external antenna, seems like pretty exciting progress to me.

Tyler-2, 2025-02-22:
I have a rural neighborhood mesh and it's all SX1276's. From a practical perspective, the T-1000's perform worse, because they can't make use of the mesh out here.

Xaositek, 2025-02-22:
> I have a rural neighborhood mesh and it's all SX1276's. From a practical perspective, the T-1000's perform worse, because they can't make use of the mesh out here.

If your T1000 is used in certain areas only, drop in an RAK4631 on USB power and let it bridge for you. While it’s not a popular answer I think there is hope in Version 3.0 to reinvest some time into seeing about creating compatibility between these radios. It’s a long shot but out there

garthvh, 2025-02-22:
Since the main feature of 3.0 is about to be released in 2.6, 3.0 is back to being 18 months away minimum, none of my 1276 devices lasted that long so this problem is likely to solve itself by the time 3.0 is released as there will be few devices with a then 15 year old radio in the wild..

egzumer, 2025-02-22:
The issue is just with a none standard syncword that meshtatstic uses. It works just fine with default syncword, so this is not a case of lack of backwards compatibility caused by some super duper new feature that makes communication 100x better, and semtech says fu, we are not going to fix it.
On the other hand, the documentation specifically says not to use non standard syncwords.

GUVWAF, 2025-02-22:
Yes, it's true it can be fixed when changing to the "private network" sync word, but that would be a breaking change.

> On the other hand, the documentation specifically says not to use non standard syncwords.

It's not mentioned anywhere in the SX127x datasheet and for the SX126x it mentions the two for public and private network, but not that you shouldn't use non-standard sync words. Meshtastic chose this about 5 years ago, and apparently it was an unfortunate choice.

gustavoulyssea, 2025-03-11:
I am going through this issue and am I looking forward to hearing some good news.

ab0oo, 2025-03-12:
FWIW, the choice of a non-standard syncword means that you can't use Semtech's really nice, high-end LoRa stuff like their SX1303 baseband chips, which are found in RAK5146 gateway boards.  Semtech allows you to use 0x12 for private and 0x34 for public, and [NOTHING ELSE](https://github.com/Lora-net/sx1302_hal/blob/4b42025d1751e04632c0b04160e0d29dbbb222a5/libloragw/src/loragw_sx1302.c)
IMHO, it would be super cool to have a multi-channel gateway in an RF advantaged location that listens to all 8 meshtastic channels simultaneously, and even allowed for cross-channel comms.
But hey, using 0x2b, we leave open the option for further "Not 2b" code/puns (per the source code comments)

garthvh, 2025-03-12:
> FWIW, the choice of a non-standard syncword means that you can't use Semtech's really nice, high-end LoRa stuff like their SX1303 baseband chips, which are found in RAK5146 gateway boards. Semtech [...]

No radiolib support is a much bigger issue for the 13xx

ab0oo, 2025-03-12:
> 
> No radiolib support is a much bigger issue for the 13xx

Chicken and egg. If it would support custom sync_words, it would become popular among meshtastic users, especially people deeply into the technical side that put up RF-advantaged nodes.  If it became popular, jgromes would have incentive to include it.  
Even without radiolib support, the meshtastic protocol is well-documented enough to write custom code for the PC/RasPi that would use the 130x.  But the inability to change the sync_word is a deal-killer.

garthvh, 2025-03-13:
Nothing to do with the syncword https://discord.com/channels/867578229534359593/972586878337433670/1180800977775902771

PYLimapper, 2025-03-19:
Hello,

Any update ? I have both devices (T1000E and T beam).

Thanks !

thebentern, 2025-05-23:
Calling this closed with #6653

garthvh, 2025-08-04:
3.0 will still use a custom meshtastic sync word.