meshtastic/firmware #4783: [Bug]: How to roll back firmware?

dmitrym-web, 2024-09-20:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.4.2

### Description

What version do I need to roll back to to get the display backlight working again like when I bought it?

dmitrym-web, 2024-09-20:
What do you close all the time and what version do you need?А какая это

b8b8, 2024-09-20:
display backlight has now been programmed to 4 clicks. the original implementation was 2 clicks which conflicts with another action right now.

todd-herbert, 2024-09-20:
Context for future readers: see #4718

dmitrym-web, 2024-09-20:
> Thanks, I wouldn’t have realized that’s what they were reporting based on this issue

so the developers broke the standard functionality to please the DIYers

thebentern, 2024-09-20:
> so the developers broke the standard functionality to please the DIYers

Enough nonsense... I'm closing this discussion. You still have this functionality through another path as answered above, and you can read through those issue threads about why we made this decision. It has nothing to do with DIYers.