meshtastic/firmware #4788: [Feature Request]: modifying firmware using gitpod - stable firmware access

ilker-aktuna, 2024-09-20:
### Platform

ESP32

### Description

I am trying to make a small mod to the firmware using gitpod. 
I can do it on the alpha firmware  using:
[https://gitpod.io#https://github.com/meshtastic/firmware](https://gitpod.io/#https://github.com/meshtastic/firmware)

but I want to do it on the stable firmware (2.4)
So I access using:
[https://gitpod.io#https://github.com/meshtastic/firmware/tree/2.4](https://gitpod.io/#https://github.com/meshtastic/firmware/tree/2.4)

but in this workspace, there is no "pio" to build.
Can it be added ?

Talie5in, 2024-09-20:
Try with the tags

https://github.com/meshtastic/firmware/tree/v2.4.3.91d6612

ilker-aktuna, 2024-09-20:
nope:

gitpod /workspace/firmware ((91d6612...)) $ pio
bash: pio: command not found

Nestpebble, 2024-09-20:
I _think_ if it's older than the gitpod setup files were introduced, you need to do it manually.

Why can't you open it for the latest, and then tell it to fetch the older version?