meshtastic/firmware #4796: [Bug]: t-beam won't stay shutdown

65a, 2024-09-21:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.5.0.9ac0e26

### Description

Recent lilygo t-beam v1.x

Shutting down the device via user long-press or android app works, but after 30s, the t-beam board will boot up again. It appears impossible to force the board off without removing the 18650.

### Relevant log output

```Shell
n/a
```

todd-herbert, 2024-09-21:
I can replicate this on T-Beam V1.2 and latest master
First guess, maybe shutdown is hanging while trying to end I2C?

```
INFO  | ??:??:?? 1453 [Button] Shutdown from long press
INFO  | ??:??:?? 1456 [Button] Shutting down
INFO  | ??:??:?? 1456 [Button] Entering deep sleep forever
INFO  | ??:??:?? 1456 [Button] Disable bluetooth until reboot
INFO  | ??:??:?? 1456 [Button] GPS deep sleep!
INFO  | ??:??:?? 1456 [Button] GPS power state moving from ACTIVE to OFF
INFO  | ??:??:?? 1457 [Button] Turning off screen
INFO  | ??:??:?? 1457 [Button] Saving /prefs/config.proto
INFO  | ??:??:?? 1457 [Button] Saving /prefs/module.proto
INFO  | ??:??:?? 1457 [Button] Saving /prefs/channels.proto
INFO  | ??:??:?? 1458 [Button] Saving /prefs/db.proto
[1458821][E][Wire.cpp:526] write(): NULL TX buffer pointer
[1458827][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
E (1589348) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (1589348) task_wdt:  - loopTask (CPU 1)
E (1589348) task_wdt: Tasks currently running:
E (1589348) task_wdt: CPU 0: IDLE0
E (1589348) task_wdt: CPU 1: IDLE1
E (1589348) task_wdt: Aborting.

abort() was called at PC 0x401584a0 on core 0

Backtrace: 0x400842c5:0x3ffbfabc |<-CORRUPTED

ELF file SHA256: bf8e607e07926086

E (45793) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (852) esp_core_dump_flash: No core dum���ѥѥ���found!
E (852) esp_core_dump_flash: No core dump partition found!
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   474][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (16384/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/cannedConf.proto (31 Bytes)
DEBUG | ??:??:?? 0  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (170 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (396 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (116 Bytes)
[   589][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[   603][I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
[   608][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
INFO  | ??:??:?? 0 AXP2101 PMU init succeeded, using AXP2101 PMU
[   631][D][XPowersAXP2101.tpp:2791] setInterruptImpl(): DISABLE - HEX:0xffffffff BIN:
DEBUG | ??:??:?? 0 =======================================================================
DEBUG | ??:??:?? 0 DC1  : +   Voltage:3300 mV
DEBUG | ??:??:?? 0 DC2  : -   Voltage:500 mV 
DEBUG | ??:??:?? 0 DC3  : -   Voltage:500 mV
DEBUG | ??:??:?? 0 DC4  : -   Voltage:1800 mV 
DEBUG | ??:??:?? 0 ALDO1: -   Voltage:1800 mV
DEBUG | ??:??:?? 0 ALDO2: +   Voltage:3300 mV 
DEBUG | ??:??:?? 0 ALDO3: +   Voltage:3300 mV
[... 109 lines cut ...]
INFO  | ??:??:?? 1 Radio myRegion->numChannels: 52 x 250.000kHz
INFO  | ??:??:?? 1 Radio channel_num: 52
INFO  | ??:??:?? 1 Radio frequency: 927.875000
INFO  | ??:??:?? 1 Slot time: 77 msec
INFO  | ??:??:?? 1 Set radio: final power level=22
INFO  | ??:??:?? 11 SX126x init result -2
WARN  | ??:??:?? 11 Failed to find SX1262 radio
DEBUG | ??:??:?? 11 SX126xInterface(cs=18, irq=33, rst=23, busy=32)
DEBUG | ??:??:?? 11 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO  | ??:??:?? 11 Starting meshradio init...
DEBUG | ??:??:?? 11 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 11 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 11 Radio freq=927.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 11 Set radio: region=ANZ, name=LongFast, config=0, ch=51, power=30
INFO  | ??:??:?? 11 Radio myRegion->freqStart -> myRegion->freqEnd: 915.000000 -> 928.000000 (13.000000 MHz)
INFO  | ??:??:?? 11 Radio myRegion->numChannels: 52 x 250.000kHz
INFO  | ??:??:?? 11 Radio channel_num: 52
INFO  | ??:??:?? 11 Radio frequency: 433.125000
INFO  | ??:??:?? 11 Slot time: 77 msec
INFO  | ??:??:?? 11 Set radio: final power level=22
INFO  | ??:??:?? 22 SX126x init result -2
WARN  | ??:??:?? 22 Failed to find SX1268 radio
INFO  | ??:??:?? 22 Not using WIFI
ERROR | ??:??:?? 22 NOTE! Recording critical error 3 at src/main.cpp:1058
```

todd-herbert, 2024-09-21:
The issue seems to have been created by #4418, affecting releases later than v2.4.2. 
When preparing to deep-sleep, `Wire.end()` is called before the T-Beam's PMU can be configured, which requires I2C.