meshtastic/firmware #4825: [Bug]: Rak4631 corrupts config or firmware after bluetooth reconnect

richardbarkeriom, 2024-09-22:
### Category

BLE

### Hardware

Rak4631

### Firmware Version

various

### Description

I had a considerable amount of trouble with my Rak4631 with settings being wiped and sometimes the entire firmware. I had the board replaced but no better. I tried multiple firmware versions too. In the end I tracked it down to bluetooth, with disconnect and reconnect events. I have since disabled bluetooth and it has been solid as a rock with no loss of settings or firmware for some weeks now. I have tried fw between 2.3.15 and 2.4.3.91d6612. 

The corruption wouldn't happen every time and more likely after running for 2-3 days. I have completely stopped the issue by turning off bluetooth.

### Relevant log output

```Shell
Not available as config corrupted and sometimes whole firmware.
```

garthvh, 2024-09-22:
Are you on mqtt via the default channel and topic?

richardbarkeriom, 2024-09-22:
Hello,
No I do not have MQTT enabled.
Regards,
Richard Barker

i-TomCZ, 2024-09-23:
Hello, i have same issue. But i think its during forced BL disconnect. For example going out of BL range. I had FW corruptions at 2.3 requiring formating of the board. With latest 2.4.3.91... it looses config pretty much always on forced BL disconnect. Tested on 3 rak boards.
Also no MQTT.

thebentern, 2024-09-25:
I believe we may have fixed this with some better teardown of bluetooth on disconnect #4834
Closing for now. Let's reopen if we see it again.

closed 2024-09-25 by thebentern

i-TomCZ, 2024-10-22:
Sorry for late response.
Unfortunately the issue is still persistent. And new issue appeared, config/firmware corruption when the node is on charger while connected to android phone with full Node DB. When battery is fully charged and not uplugged immediately then config corruption appears and sometimes even firmware gets corrupted. Then it needs to be reflashed with factory reset and 'noproto' command in Meshtastic CLI. Version is 2.5.7  but it started doing that in previous versions.