meshtastic/firmware #4828: [Bug]: ESP32 - EVENT_LOW_BATTERY can cause devices with external (intelligent) charging controllers to sleep forever

dm5tt, 2024-09-23:
### Category

Other

### Hardware

Other

### Firmware Version

2.5.x

### Description

I'm seeing increased reports of "My device doesn't wake up after charging". Guess we have a problem in the algorithm:

**Assume following scenario**

1. We applying power to an entirely discharged battery through an intelligent charge controller (BQ25185, CP*, etc.) with dedicated LOAD pins

2. Battery Voltage slowly rises in( 2.8V->2.9V->3.0V->3.1V->3.1V->V3.15->etc.)

**We will never see USB 5.0V Voltage which gets checked at**

```
Power.cpp, 683
    if (batteryLevel && powerStatus2.getHasBattery() && !powerStatus2.getHasUSB()) {
```

3. Device is waking up around 2.8V-3.0V

4.  We get the 10 hits of "Low Battery" 

5. Boom. Deep Sleep for a day (if ROUTER) or forever. But in the background we are still in a happy state as the battery is charging.

Am I seeing this right?

**Solution Ideas**

1. Reduce Deep Sleep duration 

or

2. Add a define (#define EXTERNAL_BAD_VOLTAGE_CONTROL) that disables the EVENT_LOW_BATTERY transition if we are having an external charge controller and enabled the hardware watchdog

popiciulo, 2024-09-27:
Same problem here.  I have disabled power saving mode and still the same issue, device is not booting up even though the battery is fully charged.

mverch67, 2024-10-02:
You can (still) configure how long the deep sleep lasts (power.sds_secs) . If you set it e.g. to 30 minutes, it will wake up from deep sleep after 30 minutes and if battery is still low fall asleep again otherwise continue to work.

closed 2024-12-09 by github-actions[bot]