meshtastic/firmware #4833: [Bug]: 2.5.1 random reboots on RAK and Station G2

smarsh-biolinq, 2024-09-23:
### Category

Hardware Compatibility, BLE

### Hardware

Rak4631, Station G2

### Firmware Version

2.5.1

### Description

On both units, i completely reset the install, and it will randomly reboot. I notice it when on BLE, but i am not certain it is not happening when not connected to BLE. Had to revert back to 2.5.0

remowashere, 2024-09-23:
Same on Seeed Sensecap T1000 and T-Echo, but reboots only (seems to) happen when connected via BT (latest iOS app).

thebentern, 2024-09-23:
Feel like trying an experimental build? This resolved the issue for me.
https://github.com/meshtastic/firmware/actions/runs/10999713818

smarsh-biolinq, 2024-09-23:
> Feel like trying an experimental build? This resolved the issue for me. https://github.com/meshtastic/firmware/actions/runs/10999713818

Hi There, Thank you! i was looking at that build and was thinking of trying it as well... Thanks for the tip! BTW, aren't they all experimental builds at this point? ;-)

remowashere, 2024-09-23:
> Feel like trying an experimental build? This resolved the issue for me. https://github.com/meshtastic/firmware/actions/runs/10999713818

Can confirm, on all mentioned devices this firmware works without a problem 👍

thebentern, 2024-09-25:
Closed with #4834

closed 2024-09-25 by thebentern