meshtastic/firmware #4838: [Housekeeping]: Get PR artifacts (posted in comments) working again

thebentern, 2024-09-23:
### Platform

other

### Description

@jp-bennett mentioned this https://github.com/PicoCentauri/comment-artifact maybe?

thebentern, 2024-09-23:
Tagging @scruplelesswizard as well.

scruplelesswizard, 2024-09-23:
[Relevant Workflow](https://github.com/meshtastic/firmware/blob/c442cd7267820681a2259d5b5b4531370af2377c/.github/workflows/main_matrix.yml#L219-L229)

thebentern, 2024-09-23:
> [Relevant Workflow](https://github.com/meshtastic/firmware/blob/c442cd7267820681a2259d5b5b4531370af2377c/.github/workflows/main_matrix.yml#L219-L229)

One strategy we can look at, if we can't get an action that "does it all" like the existing one, is to break it up into a commit and push to the artifacts repo and then comment step separately, which we know works fine.

closed 2024-09-29 by thebentern