meshtastic/firmware #4855: [Bug]: RP2040 - Decrypt or Encrypt (during remote admin for exemple) crashes the device

TheMalkavien, 2024-09-24:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.5 +

### Description

Hi,

Since 2.5+, my rp2040-lora was unusable. When trying to remote admin, the devices just crash.
I finally took time to do some investigations and found out it crashes just after the "Attempting PKI decryption" message.

I found out that there was some pointer dereference on "extraNonce" causing this crash. The pointer itself is not bad (valid address) but was not word-aligned on the memory. While it may work on most hardware (with non-guaranted results), it makes rp2040 crash.

Good news is I have a fix, which you may use or not, depending on how you want to handle this situation.

Firstly, I replaced the dereferences "*extraNonce" with a memcpy, like that on 2 or 3 places : 

` *extraNonce = extraNonceTmp;    ==> memcpy(extraNonce, &extraNonceTmp, 4); // do not use dereference on non aligned pointers`

I tested, and it worked further until another crash. I again found out a lot of pointer dereference into the byte buffer, which is not very good news (you can't replace every dereference !), but can be fixed in this case with a simple alignment directive on the buffer itself during declaration (the buffer was not aligned at all !) : 

```
static uint8_t bytes[MAX_RHPACKETLEN] __attribute__ ((__aligned__(4)));
static uint8_t ScratchEncrypted[MAX_RHPACKETLEN] __attribute__ ((__aligned__(4)));
```

With these 2 tricks, my rp2040 is finally working fine on 2.5+ on decrypt AND crypt (tested on 2.5.1). 
The directive is not sufficient, because the code use as nonce the last 4 bytes of the buffer, and this buffer does not always have a size modulo 4. But you can easily change by using another way to get a nonce. 

I'm not sure I explained it right, but I can answer questions :)

Thanks !

TheMalkavien, 2024-09-24:
UPDATE : I also have some heltec devices.

I Had a PKI_FAILED message by trying to remote admin a 2.5.1 htv3, I checked the keys, everything were right.
So, I updated with my patched 2.5.1 (fix alignment issues), and now everything is working fine.

I'm not a professional developer, I may understand thing the wrong way, but it seems that my changes are fixing other issues related to encryption.

Hope this helps.

jp-bennett, 2024-09-24:
I could hug you for this. Such excellent work. Memory alignment is something we don't have to worry about much with X86 code, so I didn't even consider it here.

jp-bennett, 2024-09-24:
@TheMalkavien Do you want to go ahead and make a PR with these fixes?

TheMalkavien, 2024-09-25:
Hi, of course !
I will do that as son as I can, maybe today or tomorrow.
Anyway, this should be considered more as a quick fix, an inspiration, than a definitive way to fix all memory alignment problems. I feel that some compiler flags mat be used to align things depending on hardware, and of course, some coding rules.

PR #4867 (2024-09-26): Fix (some ?) memory alignment issues on the crypto part - resulting in crashes or strange behavior
Hi,

As discussed into the #4855 issue, here is my humble PR.

It fixes #4855 by avoiding the use of non aligned pointers. On some hardware, we do not care at all, but for some the result is not guaranteed, and it makes some hardware crash (rp2040 crashes, Heltecv3 PKI_FAILED, ...).

There is 2 main changes :
- Firstly, avoiding to dereference the extraNonce pointer, wich is pointing to a bytebuffer, at an address we cannot be sure is properly aligned (it depends on the size of the buffer, and th [... cut]
commit 4794cdb1 (2024-09-26): Fix (some ?) memory alignment issues on the crypto part - resulting in crashes or strange behavior (#4867)

* Replace multiple potentially non aligned pointer dereference (#4855)
First step to fix some Crypto crashes or strange behaviors

* Makes the two Crypto byte buffers aligned (#4855)
Fix #4855, and probably multiple Crypto problems depending on hardware

---------
closed 2024-09-26 by thebentern