meshtastic/firmware #4856: [Bug]: sensless values in TELEMETRY_APP packet

karamo, 2024-09-24:
### Category

Other

### Hardware

Not Applicable, T-Beam

### Firmware Version

2.5.1

### Description

When powered my T-Beam by USB, the TELEMETRY_APP shows a batteryLevel of 101% and a voltage of -0.00...

`{
...
  "decoded": {
    "portnum": "TELEMETRY_APP",
...
  },
...
  "deviceMetrics": {
    "batteryLevel": 101,
    "voltage": -0.0010000000474974513,
...
  }
}
`

closed 2024-09-24 by garthvh

karamo, 2024-09-24:
Sorry, it's a BUG!
JSON om MQTT shows the same issue:
´´´
{
  "channel": 0,
  "from": 2475137924,
  "hop_start": 7,
  "hops_away": 0,
  "id": 1518880219,
  "payload": {
    "air_util_tx": 1.69400000572205,
    "battery_level": 101,
    "channel_utilization": 5.42000007629395,
    "uptime_seconds": 19865,
    "voltage": -0.00100000004749745
  },
  "sender": "!93879b84",
  "timestamp": 1727189888,
  "to": 4294967295,
  "type": "telemetry"
}
´´´

garthvh, 2024-09-24:
101 indicates it is powered, and here is no voltage reading available for hole on usb