meshtastic/firmware #4884: [Feature Request]: Keep RX off after provisioning and before setting region AKA Silence Cousin Pookie

simon-muzi, 2024-09-27:
### Platform

NRF52, ESP32

### Description

I'm not sure exactly what the behavior is at the moment for a HF node, will it pick up all HF messages (i.e. 868 and 915)? But certainly it is picking up 915MHZ messages after provisioning.

Firstly from a practical perspective some of our customers are getting confused to frustrated that their new devices are arriving with messages from other nodes as they excitedly unbox their shiny new Meshtastic device. At the office you can imagine we have a fair few nodes that may be on, but usually it's actually the other nodes in our neighborhood (for a while we had one named "Cousin Pookie") that we have no control of that may be sending messages (usually automated) that get caught by our new devices between provisioning them and turning them off. Some of our customers think they are getting "used" devices and are very not happy about it. For us we already spend a lot of time wiping certain batches of devices, e.g. if they have been sitting overnight charging and pick up a bunch of messages.

Secondly from a UX perspective I think it does not make the most sense if before setting a region you are picking up messages from (at that point) unknown regions. Once you're up and running you will have nodes in your nodedb that you may never be able to reach again (i.e. in other regions)? Also if your goal is a private mesh or some other use case that is not interested in public broadcasts it's just added noise.

garthvh, 2024-09-27:
Those nodes are from US long fast

simon-muzi, 2024-09-27:
> Those nodes are from US long fast

Cool, thanks for the confirmation

jp-bennett, 2024-09-28:
Seems reasonable.

closed 2024-09-28 by caveman99