meshtastic/firmware #4909: [Bug]: Don't process RX when region unset by @caveman99 in #4895, even though Region is pre-set in userPrefs.h

lagunacomputer, 2024-09-30:
### Category

Other

### Hardware

T-Beam, DIY, Other

### Firmware Version

2.5.4 or 2.5.3a70d5ee

### Description

Hope to save someone some time if they don't get any external node/RX activity after the latest alpha is flashed, and they have preset the Region in userPrefs.h.

Environment:   Build your own firmware, download latest github code, tbeam clone DIY hardware w/ RMF95

Bug Description:  If you pre-configure userPrefs.h with "#define CONFIG_LORA_REGION_USERPREFS meshtastic_Config_LoRaConfig_RegionCode_US", then upon flashing and rebooting the node, RX does not work.

Workaround:  After flashing, goto Lora Config, temporarily change Region to something else (from US to Malaysia in my case). Save.  Reboot.  Go back to Lora Config, and change the Region back to US.  Save. Reboot.  Now RX works.

Note: the problem did not appear on my other node that has a E22 SX1962 radio.  The problem seems to only happen on the node with the RMF95/SX1972 radio

lagunacomputer, 2024-09-30:
DEBUG | ??:??:?? 25 [Router] Initial packet id 722601885
DEBUG | ??:??:?? 25 [Router] Partially randomized packet id 367991710
DEBUG | ??:??:?? 25 [Router] Enqueued local (id=0x15ef1b9e fr=0x68 to=0x68, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d404bacd priority=120)
DEBUG | ??:??:?? 25 [Router] Module 'routing' considered
DEBUG | ??:??:?? 25 [Router] Rx someone rebroadcasting for us (id=0x15ef1b9e fr=0x68 to=0x68, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d404bacd priority=120)
DEBUG | ??:??:?? 25 [Router] didn't find pending packet
DEBUG | ??:??:?? 25 [Router] Add packet record (id=0x15ef1b9e fr=0x68 to=0x68, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d404bacd priority=120)
DEBUG | ??:??:?? 25 [Router] handleReceived(REMOTE) (id=0x15ef1b9e fr=0x68 to=0x68, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d404bacd priority=120)
DEBUG | ??:??:?? 25 [Router] Module 'routing' wantsPacket=1
INFO  | ??:??:?? 25 [Router] Received routing from=0xe3a93468, id=0x15ef1b9e, portnum=5, payloadlen=2
DEBUG | ??:??:?? 25 [Router] Routing sniffing (id=0x15ef1b9e fr=0x68 to=0x68, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d404bacd priority=120)
DEBUG | ??:??:?? 25 [Router] Received an ack for 0xd404bacd, stopping retransmissions
DEBUG | ??:??:?? 25 [Router] Delivering rx packet (id=0x15ef1b9e fr=0x68 to=0x68, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d404bacd priority=120)
DEBUG | ??:??:?? 25 [Router] Update DB node 0xe3a93468, rx_time=0
DEBUG | ??:??:?? 25 [Router] Forwarding to phone (id=0x15ef1b9e fr=0x68 to=0x68, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d404bacd priority=120)
DEBUG | ??:??:?? 25 [Router] Module 'routing' considered
INFO  | ??:??:?? 30 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | ??:??:?? 30 [NodeInfoModule] sending owner !e3a93468/KM6HHP_base/KM6H
DEBUG | ??:??:?? 30 [NodeInfoModule] Partially randomized packet id 158296991
DEBUG | ??:??:?? 30 [NodeInfoModule] Update DB node 0xe3a93468, rx_time=0
DEBUG | ??:??:?? 30 [NodeInfoModule] handleReceived(LOCAL) (id=0x096f6b9f fr=0x68 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
DEBUG | ??:??:?? 30 [NodeInfoModule] Add packet record (id=0x096f6b9f fr=0x68 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] Expanding short PSK #1
DEBUG | ??:??:?? 30 [NodeInfoModule] Using AES128 key!
DEBUG | ??:??:?? 30 [NodeInfoModule] enqueuing for send (id=0x096f6b9f fr=0x68 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] txGood=0,rxGood=0,rxBad=0
DEBUG | ??:??:?? 30 [RadioIf] Starting low level send (id=0x096f6b9f fr=0x68 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=10)
DEBUG | ??:??:?? 30 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=99, time 968 ms
DEBUG | ??:??:?? 30 [RadioIf] AirTime - Packet transmitted : 968ms
DEBUG | ??:??:?? 31 [RadioIf] Completed sending (id=0x096f6b9f fr=0x68 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=10)
INFO  | ??:??:?? 32 Rebooting
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (850) esp_core_dump_flash: No core dump▒▒ѥѥ▒▒▒found!
E (850) esp_core_dump_flash: No core dump partition found!
[    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (601) psram: PSRAM ID read error: 0xffffffff
[    27][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!
▒▒@INFO  | ??:??:?? 0

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (16384/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (166 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (389 Bytes)
[   115][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=5 scl=4 freq=100000
DEBUG | ??:??:?? 0 Power::lipoInit lipo sensor is not ready yet
DEBUG | ??:??:?? 0 Using analog input 35 for battery level
INFO  | ??:??:?? 0 ADCmod: ADC characterization based on reference voltage stored in eFuse
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   155][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for I2C devices on port 1
DEBUG | ??:??:?? 0 I2C device found at address 0x3c
INFO  | ??:??:?? 0 ssd1306 display found
INFO  | ??:??:?? 0 ssd1306 display found
DEBUG | ??:??:?? 0 0x6 subtype probed in 2 tries
INFO  | ??:??:?? 0 1 I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 S:B:255,2.5.4.d73cbf14
[... 749 lines cut ...]
DEBUG | ??:??:?? 200 [Power] Battery: usbPower=0, isCharging=0, batMv=4014, batPct=84
DEBUG | ??:??:?? 220 [Power] Battery: usbPower=0, isCharging=0, batMv=4012, batPct=83
INFO  | ??:??:?? 225 [DeviceTelemetryModule] (Sending): air_util_tx=0.069528, channel_utilization=0.000000, battery_level=83, voltage=4.012000, uptime=225
DEBUG | ??:??:?? 225 [DeviceTelemetryModule] Partially randomized packet id 92412716
DEBUG | ??:??:?? 225 [DeviceTelemetryModule] updateTelemetry LOCAL
DEBUG | ??:??:?? 226 [DeviceTelemetryModule] Node status update: 3 online, 3 total
[... cut ...]
INFO  | ??:??:?? 286 [DeviceTelemetryModule] Sending packet to phone
INFO  | ??:??:?? 286 Telling client we have new packets 11
INFO  | ??:??:?? 286 BLE notify fromNum
INFO  | ??:??:?? 286 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 286 phone downloaded packet (id=0xf0984f2d fr=0x68 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=10)
DEBUG | ??:??:?? 301 [Power] Battery: usbPower=0, isCharging=0, batMv=4015, batPct=84
DEBUG | ??:??:?? 321 [Power] Battery: usbPower=0, isCharging=0, batMv=4019, batPct=84

caveman99, 2024-09-30:
This is a red herring and has nothing to do with the patch, or it would have happened on the SX1262 too. Also there would be log entries alerting you to the RX suppression. The SX1276 still seem to have problems with the recent radiolib update, that's why the older versions worked. I'll investigate

caveman99, 2024-09-30:
@GUVWAF any idea from the top of your head? we're de facto on radiolib 7.0.1

GUVWAF, 2024-09-30:
Strange, it looks like a different DIO mapping is applied now: 
https://github.com/jgromes/RadioLib/commit/38fc7a97a4c195b7c10aa94215a1c53ec18a56ef#diff-00e5593e444a00c69ace49442b07515a9fecbf542ce7288a558bf5a0f3d8ce52R1403 
Shouldn’t this be the other way around: `(dioNum <= 3)`?

caveman99, 2024-09-30:
yes, that really looks wrong RADIOLIB_SX127X_REG_DIO_MAPPING_2 is only used for DIO4 and bigger. either <=3 or swap the labels.

caveman99, 2024-09-30:
i opened https://github.com/jgromes/RadioLib/issues/1244 - seems like a quick fix.

caveman99, 2024-09-30:
I just confirmed that patch fixes the Problem. Waiting for Radiolib 7.0.2 as a hotfix

closed 2024-09-30 by caveman99