meshtastic/firmware #4914: [Feature Request]: Request Key Exchange with user known Node Hex ID

Coopersmith-24601, 2024-09-30:
### Platform

Cross-Platform

### Description

There are times, notably when updating firmware or other node db wipes, when User A needs to be able to contact another User B.  However, User B may not be in User A's node db yet.  User B may also have a relatively long period set for broadcasting meaning it may not show up for an extended time.

It would be useful if Node A could force a broadcast _and key exchange_ of Node B by entering in Node B's specific Hex ID.

caveman99, 2024-09-30:
If node A does not have the key for B it will send the DM 'old style'. Also if Node B can not decrypt a DM becasue it has the wrong key, it sends its own nodeinfo (including its key) asking for reply.

Coopersmith-24601, 2024-10-01:
Right, but you CAN'T send a DM to a node which isn't in the Node DB yet.

What I'm suggesting is a user be able to input the HEX ID of a node they know which then also pings the node and requests it do a key exchange so users don't have to wait.