meshtastic/firmware #4920: [Bug]: unable to disable serial output

boxaroost, 2024-10-01:
### Category

Serial

### Hardware

T-Deck, T-Echo

### Firmware Version

2.5.3.a70d5ee

### Description

After switching to 2.5 alpha (2.5.3.a70d5ee), I found I cannot disable the info logging over serial. On Android I can change the Device-> serial output enabled toggle, and it stays changed, however in either state the logging continues.  Viewing the serial on Windows 11

boxaroost, 2024-10-01:
I use my connected node for caltopo integration, the extra logging makes
the caltopo console very busy. As turning the console info off is useful,
why is it not intended to fix the use of this toggle? (And isn't that still
a bug to be fixed?)

thebentern, 2024-10-02:
> I use my connected node for caltopo integration, the extra logging makes the caltopo console very busy. As turning the console info off is useful, why is it not intended to fix the use of this toggle? (And isn't that still a bug to be fixed?)

Would it be helpful to turn most (if not all) of the other serial logging off if we're in CalTopo mode over console serial port?

boxaroost, 2024-10-02:
It could be useful for debug, the most optimal is the toggle working like
it does in 2.4 (in caltopo mode the toggle still turns the logging on/off).
I'm not sure what's more useful for users, the extra logging or no logging,
probably users in both camps.

fifieldt, 2024-11-16:
Q) what role is your device? Router?

garthvh, 2024-11-16:
The settings under security config now, works in my testing, are you using the new setting?

closed 2025-01-23 by github-actions[bot]