meshtastic/firmware #4929: [Bug]: T-Deck PLUS GPS does not consistently work on FW 2.5.1 as received by Lilygo

captainchaosrules, 2024-10-02:
### Contact Details

captainchaosrules@gmail.com

### What happened?

Hi,

If the GPS is not wired or configured correctly it will show “No GPS and “GPS not present”
When the GPS is working correctly and indoors for example it will say” No Sats and No GPS lock"
Switching these T-Deck Plus’s on and off sometimes the GPS works sometimes it does not.

Has this fix been implemented in FW 2.5.1?
https://github.com/meshtastic/firmware/issues/4625
  

### App Version

2.4.4

### Phone

Pixel 6 

### Device

T-Deck PLUS

### Firmware

2.5.1

### Relevant log output

```shell
n/a
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

LoganRickert, 2024-10-03:
Hello,

I also experienced this issue (though I never went outside with it while on 2.5.1).

I updated to to 2.5.4 (I had to hold in the trackball) and after going outside for a bit and coming back in, I had a GPS signal while I was outside the whole time. I have a GPS signal while indoors after rebooting each time (while on 2.5.4).

fifieldt, 2024-10-12:
Fixed by: https://github.com/meshtastic/firmware/pull/5014

closed 2024-10-12 by fifieldt