meshtastic/firmware #4939: [Bug]: T-Deck Plus Screen Resets every 30 seconds

LoganRickert, 2024-10-03:
### Category

Hardware Compatibility

### Hardware

T-Deck

### Firmware Version

2.5.4.8d288d5 Alpha

### Description

I am using a T-Deck Plus. Almost exactly every 30 seconds, the screen will turn off and back on and go back to the 'home' screen. This is annoying because if I am typing a message, the message I was typing is erased from the screen.

If I change the value of `Light Sleep Interval` from 30 to 300, the screen does not flash off and back on but the screen is still reset every 30 seconds.

Here is the log from the serial debug console:

```
[32mINFO  [0m| 20:31:05 201 [PowerFSM] [32mTurning off screen
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34maction LONG PRESS(0/24693)
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mEnter state: ON
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mshowing standard frames
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mShowing 0 module frames
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mTotal frame count: 103
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mAdded modules.  numframes: 0
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mFinished building frames. numframes: 5
[0m[32mINFO  [0m| 20:31:05 202 [Screen] [32mTurning on screen
[0m[34mDEBUG [0m| 20:31:05 202 [Screen] [34mBrightness is set to value: 130
```

This happen while both connected to USB and not connected to USB. This happens while typing and interacting and doing nothing.

I let it run for a few minutes and copied every time action LONG PRESS appears

```
16:49:34.667 -> [0m[34mDEBUG [0m| 20:49:34 384 [touchscreen1] [34maction LONG PRESS(0/24693)
16:49:54.682 -> [0m[34mDEBUG [0m| 20:50:05 414 [touchscreen1] [34maction LONG PRESS(0/24693)
16:50:34.700 -> [0m[34mDEBUG [0m| 20:50:35 444 [touchscreen1] [34maction LONG PRESS(0/24693)
16:50:54.717 -> [0m[34mDEBUG [0m| 20:51:05 475 [touchscreen1] [34maction LONG PRESS(0/24693)
16:51:34.710 -> [0m[34mDEBUG [0m| 20:51:36 505 [touchscreen1] [34maction LONG PRESS(0/24693)
16:51:54.715 -> [0m[34mDEBUG [0m| 20:52:06 536 [touchscreen1] [34maction LONG PRESS(0/24693)
```

You can observe that it appears to happen almost exactly every 30 seconds.

Thank you.

### Relevant log output

```Shell
[32mINFO  [0m| 20:31:05 201 [PowerFSM] [32mTurning off screen
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34maction LONG PRESS(0/24693)
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mEnter state: ON
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mshowing standard frames
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mShowing 0 module frames
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mTotal frame count: 103
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mAdded modules.  numframes: 0
[0m[34mDEBUG [0m| 20:31:05 202 [touchscreen1] [34mFinished building frames. numframes: 5
[0m[32mINFO  [0m| 20:31:05 202 [Screen] [32mTurning on screen
[0m[34mDEBUG [0m| 20:31:05 202 [Screen] [34mBrightness is set to value: 130
```

LoganRickert, 2024-10-03:
Spoke with my friend and his touch screen works, mine does not. It’s very possible that my touch screen is somehow broken from the factory. We’re going to try and reseat the ribbon cable and see if this fixes the issue.

369Martin369, 2024-10-07:
wifi related?

BuddyBadger, 2024-11-06:
I was having the exact same issue with my T-Deck Plus. Turns out, I accidentally left the factory screen protector on the touch display. I removed it and mine is now working fine. 🤦‍♂️

queekusme, 2024-12-09:
Mine did this, I went into Display > Screen On For, cycled it to another then back to 30 and it started working as expected...

My factory screen protector is still on

joyel24, 2024-12-15:
@LoganRickert Can you reproduce this bug in 2.5.17 ?
I give you this bin wich is from latest commit of tft-gui-work + some french translation I just carried out.
[firmware.bin.zip](https://github.com/user-attachments/files/18142300/firmware.bin.zip)

queekusme, 2024-12-15:
> @LoganRickert Can you reproduce this bug in 2.5.17 ?
> 
> I give you this bin, you won't need to build it.
> 
> it's from latest commit of tft-gui-work + some french translation I just carried out.
> 
> [firmware.bin.zip](https://github.com/user-attachments/files/18142300/firmware.bin.zip)
> 
> 

Mine stopped doing it in *.16

joyel24, 2024-12-15:
@queekusme Thanks for this clarification.

Si I guess this issue should be closed...

LoganRickert, 2025-01-16:
@joyel24  Apparently GitHub does not send me any notifications for this thread lol. I am currently away but I will test this next week and see what happens.

nasiralamreeki, 2025-02-27:
> @joyel24  Apparently GitHub does not send me any notifications for this thread lol. I am currently away but I will test this next week and see what happens. 

Any update?

fifieldt, 2025-02-27:
There is a new firmware preview for T-Deck that is a significant improvement, I recommend trying it out: https://meshtastic.org/blog/meshtastic-2-6-preview/ !

closed 2025-02-27 by thebentern