meshtastic/firmware #4964: [Feature Request]: Make it easier to add a single channel

K1WIZ, 2024-10-05:
### Platform

Cross-Platform

### Description

Channel management continues to be frustrating.  It is not intuitive nor easy to add a single channel to an existing channel set.   This makes it hard to join other meshes without a lot of CLI/QR/URL  trickery.   Please consider implementing an easy function to add a new channel (without replacing existing ones) via the iOS or android app.  It's a very clunky experience.

thebentern, 2024-10-05:
Good news! This is actually already available on iOS. I assume it's either in Android already as well or will be soon

https://github.com/user-attachments/assets/e243e883-3928-42fa-95b3-d0033ff22e6a

thebentern, 2024-10-05:
This notably generates a QR code / URL with a `add=true` query string which is the queue to clients to add instead of replace.:
`https://meshtastic.org/e/?add=true#.............`

Also worth mentioning, you can just take an existing link and add `?add=true` in front of the # and make it an add only link. 😄

closed 2024-10-05 by thebentern

Nikotine1, 2025-04-16:
> Good news! This is actually already available on iOS. I assume it's either in Android already as well or will be soon
> ScreenRecording_10-05-2024_11-45-40_1.mov

This isn't possible yet with the Android app I believe?