meshtastic/firmware #4978: [Bug]: segmentation fault for armhf on pi zero

tahnok, 2024-10-07:
### Category

Other

### Hardware

Linux Native

### Firmware Version

2.5.5.e182a

### Description

Launching `meshtasticd` on the pi zero w  (rev 1.1) results in a segmentation fault. The issue also occurs on version `2.5.4.8d288d5`

### Relevant log output

```Shell
gdb backtrace has 

(gdb) bt
#0  0x00000000 in ?? ()
#1  0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0xb6fdb770 <call_init+148>) at /usr/include/c++/12/bits/stl_iterator.h:1159
#2  0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0x0) at /usr/include/c++/12/bits/stl_iterator.h:1159
#3  0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0xbefff554) at /usr/include/c++/12/bits/stl_iterator.h:1159
#4  0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (
    this=0x4664f5 <std::allocator_traits<std::allocator<graphics::Screen::ScreenCmd> >::max_size(std::allocator<graphics::Screen::ScreenCmd> const&)+14>) at /usr/include/c++/12/bits/stl_iterator.h:1159
#5  0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0xbefff584) at /usr/include/c++/12/bits/stl_iterator.h:1159
#6  0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0x0) at /usr/include/c++/12/bits/stl_iterator.h:1159
#7  0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0xbefff564) at /usr/include/c++/12/bits/stl_iterator.h:1159
#8  0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (
    this=0x466277 <std::deque<graphics::Screen::ScreenCmd, std::allocator<graphics::Screen::ScreenCmd> >::_S_max_size(std::allocator<graphics::Screen::ScreenCmd> const&)+20>) at /usr/include/c++/12/bits/stl_iterator.h:1159
#9  0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0x0) at /usr/include/c++/12/bits/stl_iterator.h:1159
#10 0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0x0) at /usr/include/c++/12/bits/stl_iterator.h:1159
#11 0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0x0) at /usr/include/c++/12/bits/stl_iterator.h:1159
#12 0x004d5d1a in __gnu_cxx::__normal_iterator<_meshtastic_Neighbor*, std::vector<_meshtastic_Neighbor, std::allocator<_meshtastic_Neighbor> > >::base (this=0x38) at /usr/include/c++/12/bits/stl_iterator.h:1159
```

(this appears to loop forever)

registers:

```
(gdb) info registers
r0             0xb6fdb770          3070080880
r1             0x1                 1
r2             0xbefff584          3204445572
r3             0xb6fdb770          3070080880
r4             0x27f50             163664
r5             0x0                 0
r6             0x416dc1            4287937
r7             0xb6fdb770          3070080880
r8             0x0                 0
r9             0x0                 0
r10            0xb6fff000          3070226432
r11            0x0                 0
r12            0xb6fc4d38          3069988152
sp             0xbefff54c          0xbefff54c
lr             0x4d5d1b            5070107
pc             0x0                 0x0
cpsr           0x80000010          -2147483632
fpscr          0x0                 0
```
```

jp-bennett, 2024-10-07:
This is almost certainly a consequence of building 32 bit binaries in a 32 bit docker image on a 64 bit install. Just needs a 32 bit builder dedicated to the task.

rjbrown99, 2024-10-07:
This impacts me as well, including with the latest 2.5.5.e182ae7 release on an RPi Zero 2W. Latest Raspbian distro with 6.6.46-1+rpt1 kernel.

jp-bennett, 2024-10-12:
I haven't nailed down exactly what causes this, but it's a problem with how we build the 32-bit binaries. Hopefully will get this working again soon.

rjbrown99, 2024-10-13:
Lazy question: does anyone know the most recent release that does work on the RPi Zero? 

I tried building it myself but ran into issues with the toolchain and platformio.

`/home/foo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ar: Failed to load plugin '/root/.platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so', reason: /root/.platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc`

If not I'll go back through the releases and see if I can find one that works.

jp-bennett, 2024-10-13:
You need to specify that you want to use the native environment. I think it's "-e native" in the pio command.

jp-bennett, 2024-11-10:
The 32 bit arm build from https://github.com/meshtastic/firmware/actions/runs/11760255439?pr=5294 may work.

roens, 2024-11-10:
> The 32 bit arm build from https://github.com/meshtastic/firmware/actions/runs/11760255439?pr=5294 may work.

I can confirm: it does not segfault! It did exit with a return code of 134, but that's probably because of my config. I might be able to do more testing tomorrow.

tahnok, 2024-11-13:
I can confirm it installs and starts on a Pi, but I'm still struggling to transmit or receive anything (but that may be user error)

broglep, 2024-11-13:
I can confirm that build from PR works fine so far on Raspberry Pi Zero. Had previously resorted to running in docker with `grinco/meshtasticd:armv6-2.5.5` and now have switching over to meshtasticd_2.5.12.6c536ce, no segfaults so far and send/receive also works as expected

tahnok, 2024-11-14:
After adding some bodge wires, I can receive and transmit on a pi zero using an rfm9x module with the linked build

jp-bennett, 2024-11-18:
I think we have this one fixed then. We were doing ARMv7 builds, and those don't work on the ARMv6 Pis.

closed 2024-11-18 by jp-bennett