meshtastic/firmware #4980: [Bug]: T-Watch S3 shows GMT instead of GMT+2h (where I am)

369Martin369, 2024-10-07:
### Category

Other

### Hardware

T-Watch

### Firmware Version

2.5.5.e18 (alpha) and 2.5.4.8d (stable)

### Description

watch is 2h behind real time

### Relevant log output

```Shell
none
```

fifieldt, 2024-10-07:
Potentially fixed by https://github.com/meshtastic/firmware/pull/4979

thebentern, 2024-10-07:
This can also be set in the app currently over BLE pairing

369Martin369, 2024-10-07:
solved now ... (and not with above info, but YT channel Owner https://www.youtube.com/@TechAirSpace helped me :-)

In Android app (not availibe in web) you need to set in "Radio Configuration" -> "Device" -> "POSIX Timezone" (Last data entry field on bottom) a string listed in https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv

For Berlin e.g. you have to set it to "CET-1CEST,M3.5.0,M10.5.0/3"