meshtastic/firmware #4981: [Bug]: T-Wash is looping after Wifi enabled

369Martin369, 2024-10-07:
### Category

WiFi

### Hardware

T-Watch

### Firmware Version

2.5.5.e18 (alpha) and 2.5.4.8d (stable)

### Description

watch is looping and cannot enable wifi. tried to enable wifi to get time synced (other bug), but wifi does not work

### Relevant log output

```Shell
none
```

caveman99, 2024-10-07:
Please pst a serial log from the watch starting up.

369Martin369, 2024-10-08:
please explain me more how I can do that? Thanks @caveman99

caveman99, 2024-10-08:
connect an usb cable to your PC and the watch charging port, open up a command line and start `meshtastic --noproto`. The MEshtastic CLI needs to be installed like outlined in our documentation.

caveman99, 2024-10-10:
Patch doesn't fix the issue unfortunately. it now crashes in another spot.

369Martin369, 2024-10-10:
sorry, buy watch does no more work.... :-( cannot test.

todd-herbert, 2024-10-11:
Logs reported via Discord:
> caveman99 🇩🇪 — 2024/10/10 at 10:56 PM:
>
> DEBUG | 09:42:12 16 WiFi-Event 5: Disconnected from WiFi access point
> WARN  | 09:42:13 16 WiFi lost connection. Reason: 2
> INFO  | 09:42:14 17 [WifiConnect] Reconnecting to WiFi access point HomeNet
> DEBUG | 09:42:19 23 WiFi-Event 4: Connected to access point
> DEBUG | 09:42:19 23 WiFi-Event 7: Obtained IP address: 192.168.101.11
> INFO  | 09:42:19 23 Starting WiFi network services
> INFO  | 09:42:19 23 mDNS responder started
> INFO  | 09:42:19 23 mDNS Host: Meshtastic.local
> INFO  | 09:42:19 23 Starting NTP time client
> DEBUG | 09:42:19 23 Initializing Web Server ...
> INFO  | 09:42:19 23 Starting Secure Web Server...
> INFO  | 09:42:19 23 Starting Insecure Web Server...
> INFO  | 09:42:19 23 Web Servers Ready! :-) 
> 
> next dead stop

___

I was able to reproduce the first stage of the issue with my DIY Touch-TFT node, but I can't reproduce this second stage.

That DIY device uses the older ESP32 rather than ESP32-S3. As a further experiment, I built 2.5.7.1f2d972 (current master) for Heltec Wireless Paper (an S3 device), modified to show the T-Watch screen frames. I couldn't reproduce the issue with this set-up either.

___

Is T-Watch S3 expected to receive an LVGL device-ui in the future? If that is the case, it might not be super important to dig too deep into fixing this one with the old screen implementation?

Guytantak, 2024-11-12:
On mine with same bootloop after wifi enabling issue, I can't reach a com port to wipe memory. Perhaps with an USB uart converter ? But is there TX RX combo somewhere in the S3 watch inside ?

matthewbullweb, 2024-12-18:
Hi, found away to recover a bricked t3 device on com4. Use the ardunio ide on com4 using esp32-s3 dev board in usb ota mode. A empty stetch with work and watch reboots in com3 mode. You can upload via hardware tag as normal after. No more have to open the watch up to find the boot button inside.

Guytantak, 2024-12-19:
Great ! Thanks ☺️

matthewbullweb, 2025-08-11:
> Can you test against Meshtastic 2.7.1 to determine if this issue has now been resolved?

Was fixed of 2.6.* and still works on 2.7.1

anemes, 2025-09-27:
Still experiencing boot loop with Wi-Fi enabled for fresh flash of 2.6.11 beta

PR #5015 (2024-10-10): Fix bootloop for T-Watch S3 when using WiFi
Fixes #4981 (hopefully)

I can reproduce and solve the issue on my makeshift T-Watch test device.
It'd be good to get this checked on the actual hardware just to confirm.