meshtastic/firmware #4999: [Feature Request]: Change out lewisxhe/PCF8563_Library

ellisgl, 2024-10-08:
### Platform

ESP32

### Description

I see that there has been more active development on one of it's forks: https://github.com/CDFER/pcf8563-RTC

ellisgl, 2024-10-11:
Assign it to me - I'll get it done. =)

ellisgl, 2024-10-11:
https://github.com/meshtastic/firmware/pull/5031

caveman99, 2024-10-11:
From the library readme:

-     Not all functions are implemented
-     Not compatible with Lewis' library
-     Only tested with the ESP32 and S2 variants

We use this with all kinds of processor arches, and the way you built up the code request you assume it's a drop in replacement, which it is not. Alone the change from local zone to UTC requires code changes in meshtastic. I am not in favor of doing this.