meshtastic/firmware #5002: [Bug]: T-Watch S3 clock is inaccurate

thefoeyouknow, 2024-10-09:
### Category

Other

### Hardware

T-Watch

### Firmware Version

v2.5.5.e182ae7

### Description

Clocks on t-watch are inaccurate, gaining roughly 1 minute per hour, requiring BLE connection to re-sync after several hours. In 14 hours of operation, the clock can be off 12 to 16 minutes.

CcRup, 2024-12-04:
Hello,

New to GitHub so not certain how to interpret certain tags yet. Has this bug fix been corrected or not a fix we can expect? I bought a T-watch S3 to use with Meshtastic and I am experiencing the same issue. It's unusable as a watch. How can we re-open this bug fix request? Thank you.

CcRup, 2024-12-05:
Thanks again Garth. For those interested, I found that leaving power saving mode off will solve the problem in my case (to within 1-2 seconds of NIST server time). This coincides with the BT phone sync countering any irregularities from network clock sources. Best wishes!