meshtastic/firmware #5009: Not activating QZSS?

allanmac, 2024-10-09:
IThe UC6580 will accept "$CFGSYS,h35155\r\n".

This also enables QZSS reception which is visible from region around Japan as well as the Pacific CONUS.

Maybe there was a reason not to enable it?

https://github.com/meshtastic/firmware/blob/ddd4a45bc3a9d8e227f532d1a04a5508ecf54e31/src/gps/GPS.cpp#L502

Thanks!

fifieldt, 2024-10-10:
Sounds like a good idea. Testing on heltec-wireless-tracker.

fifieldt, 2024-10-10:
Seems happy enough:

```
$GNTXT,01,01,00,CFGSYS,h35155*39#

$OK*04#
```

fifieldt, 2024-10-10:
Hi @allanmac ,I put in a quick patch. If you have any other GPS settings to share they would be much appreciated!

closed 2024-10-10 by caveman99

allanmac, 2024-10-10:
Thanks!

I'll add some background here for future reference...

There is a brief note on the Heltec Wireless Tracker v1.1 site and the schematics appear to confirm that you can power up the UC6580 module (via the VEXT pin) with a USB<>TTL serial cable while leaving the Heltec board powered off.

I'm using a cable with **3.3V** out.  As far as I can tell this is safe but I may have missed something.

Connect:

- VCC (3.3V) from the cable to the board's VEXT (GPIO3/PIN7)
- GND to the board's GND (PIN8)
- RX to the board's GNSS_TX pin (PIN33)
- TX to the board's GNSS_RX pin (PIN34)

The Heltec doesn't power up but the UC6580 GPS module starts spraying NMEA at 115200/8N1.

[image]

I then looked at the GPS module using Unicom's UPrecise tool and it looks like its defaults are fine.

I went through each command and setting in the manual and everything that seems important is enabled by default -- I didn't see any disabled features.

However, the new `$CFGSYS` bitmask setting in the [spec on the Unicom site](https://en.unicore.com/uploads/file/UFB2%20Protocol/UFBII%20Protocol%20Specification%20EN/index.html) (login required) disagrees with the directly downloadable [UFirebird II Series Protocol Specification](https://en.unicore.com/uploads/file/ufirebirdii%20series%20protocol%20specification-en-r1-1.pdf) doc.  

Specifically, The QZSS and SBAS bits conflict between the two docs and an error will be returned if you try to enable those constellations with the newer bitmask.  

Additionally, the built-in antenna is only OK indoors. It does not appear to be an L1/L5 patch antenna (also confirmed by its shape).

An inexpensive external Waveshare L1/L5 patch antenna takes a while to cold start but it demonstrated that the UC6580I receiver is quite capable and even from indoors I’m seeing 70 sats and using 25 (L1+L5).

Open questions:  I'm still wondering why Meshtastic (2.5.5)  on the Heltec Wireless Tracker v1.1 periodically reports no satellites despite being connected to an external antenna. 

Is the module being powered down and effectively cold-started?

mkgin, 2024-10-10:
There's +3.3V from the main regulator through a diode to +Vbackup on the UC6580. Enough for memory, internal clocks + XTAL. There's another diode to a test point that looks like it could be used for an extra backup battery, like a CR2032 for example.

Regarding 0 satellites on start. Could it be that the firmware doesn't wait long enough for a warm start?

I didn't realize this was a multi and receiver until now. :-)