meshtastic/firmware #5010: [Bug]: PKI based node administration does not work

brekovcan, 2024-10-09:
### Category

Other

### Hardware

T-Beam, T-Echo, Rak4631

### Firmware Version

2.5.4.8d288d5

### Description

I have updated all nodes to 2.5.4.8d288d5.  I can send encrypted direct messages and encrypted message to admin channel as well.  On IOS App Settings i have selected PKI based Administration.  I have configured also Admin key on each node to be public key of node from i will do remote node administration. If i select remote node and go to any setting, like LoRA,  i can see message, There has been no response… and it is not possible to scroll.  Exiting and returning back to Lora configuration does not help.  Before upgrade from 2.4, Legacy node configuration was working.

pdxlocations, 2024-10-09:
Have you enabled Administration in the App Settings?
[image]

pdxlocations, 2024-10-09:
You no longer need an Admin Channel when using PKI based Admin.

I have noticed some odd behavior with admin:
If the admin request fails for any reason (ie, I've attempted to admin with legacy before enabling pki admin in settings) the only way I've found to try again is to Clear App Data.

closed 2024-10-09 by thebentern

brekovcan, 2024-10-09:
I have deleted IOS app, install it again, setup App Settings to use PKI based node administration, selected node, same issue, reinstalling app on IOS did not help. So what to do, delete admin channel on each node, than PKI based node administration will start working?

pdxlocations, 2024-10-09:
leaving the admin channel shouldn't hurt, it's just not necessary, in my experience.  I've been using PKI Admin with iOS successfully, perhaps you have a typo or extra space in the admin key field?

"forgetting" the remote node on the admin node is something I'd try (then wait for it repopulate), then a full erase and reinstall on both nodes to make sure everything is default.  

Be sure to backup your public an private keys if you need them before a full erase.

brekovcan, 2024-10-09:
I have discovered, that PKI node administration works only if i renamed admin channel (or delete). So in my opinion is, that in 2.5 firmware it should be somehow disabled by default  if default option is use PKI for node administration. So today i was on roof to upgrade firmware to 2.5 and now there is only one option how to enable remote administration of this node - go again on roof and rename/delete admin channel.. (bluetooth is not enabled on my node).   btw.. is it any option on IOS app to add more than one Admin ley?

pdxlocations, 2024-10-09:
Interesting, I have several remote installs running old firmware.  I'm able to admin them using the admin channel on 2.5 nodes.