meshtastic/firmware #5017: [Bug]: Station G2 won't boot after erase and flash 2.5.6

liquidraver, 2024-10-10:
### Category

Other

### Hardware

Station G2

### Firmware Version

2.5.6.ad8747d

### Description

Tried to flash the newest alpha, but the device won't boot. Flashed back 2.5.5 and that is working.

### Relevant log output

```Shell
//\ E S H T /\ S T / C

 INFO  | ??:??:?? 5 Booted, wake cause 0 (boot count 1), reset_reason=reset
 DEBUG | ??:??:?? 5 Filesystem files (1048576/1048576 Bytes):
 DEBUG | ??:??:?? 5  /static/.gitkeep (0 Bytes)
 DEBUG | ??:??:?? 5  /static/Logo_Black.svg.gz (602 Bytes)
 DEBUG | ??:??:?? 5  /static/Logo_White.svg.gz (610 Bytes)
 DEBUG | ??:??:?? 5  /static/apple-touch-icon.png.gz (3164 Bytes)
 DEBUG | ??:??:?? 5  /static/favicon.ico.gz (2270 Bytes)
 DEBUG | ??:??:?? 5  /static/icon.svg.gz (852 Bytes)
 DEBUG | ??:??:?? 5  /static/index-b9vwykds.js.gz (222120 Bytes)
 DEBUG | ??:??:?? 5  /static/index-wV0rbIjU.css.gz (16573 Bytes)
 DEBUG | ??:??:?? 5  /static/index.html.gz (547 Bytes)
 DEBUG | ??:??:?? 5  /static/mapbox-gl-DY02hNYD.js.gz (391221 Bytes)
 DEBUG | ??:??:?? 5  /static/maplibre-gl-H6mDd47p.js.gz (210429 Bytes)
 DEBUG | ??:??:?? 5  /static/robots.txt.gz (42 Bytes)
 DEBUG | ??:??:?? 5  /static/site.webmanifest.gz (197 Bytes)
 DEBUG | ??:??:?? 5 Power::lipoInit lipo sensor is not ready yet
 INFO  | ??:??:?? 5 Scanning for i2c devices...
 DEBUG | ??:??:?? 5 Scanning for I2C devices on port 1
 DEBUG | ??:??:?? 5 I2C device found at address 0x3c
 DEBUG | ??:??:?? 5 0x5 subtype probed in 2 tries 
 INFO  | ??:??:?? 5 1 I2C devices found
 DEBUG | ??:??:?? 5 acc_info = 0
 INFO  | ??:??:?? 5 S:B:31,2.5.6.ad8747d
 INFO  | ??:??:?? 5 Build timestamp: 1728516504
 DEBUG | ??:??:?? 5 Total heap: 275368
 DEBUG | ??:??:?? 5 Free heap: 233208
 DEBUG | ??:??:?? 5 Total PSRAM: 8386295
 DEBUG | ??:??:?? 5 Free PSRAM: 8386295
 DEBUG | ??:??:?? 5 NVS: UsedEntries 10, FreeEntries 620, AllEntries 630, NameSpaces 2
 DEBUG | ??:??:?? 5 Setup Preferences in Flash Storage
 DEBUG | ??:??:?? 5 Number of Device Reboots: 7
 INFO  | ??:??:?? 5 OTA firmware version 0.3.0.87f0f2f
 INFO  | ??:??:?? 5 Initializing NodeDB
 ERROR | ??:??:?? 5 Could not open / read /prefs/db.proto
 WARN  | ??:??:?? 5 Devicestate 0 is old, discarding
 INFO  | ??:??:?? 5 Installing default DeviceState
 DEBUG | ??:??:?? 5 Initial packet id 1496061209
 DEBUG | ??:??:?? 5 Partially randomized packet id 2367825178
 DEBUG | ??:??:?? 5 Using nodenum 0xa20a1890 
 ERROR | ??:??:?? 5 Could not open / read /prefs/config.proto
 INFO  | ??:??:?? 5 Installing default LocalConfig
 INFO  | ??:??:?? 5 Setting default channel and radio preferences!
 DEBUG | ??:??:?? 5 Expanding short PSK #1
 INFO  | ??:??:?? 5 Wanted region 0, using UNSET
 ERROR | ??:??:?? 5 Could not open / read /prefs/module.proto
 INFO  | ??:??:?? 5 Installing default ModuleConfig
 ERROR | ??:??:?? 5 Could not open / read /prefs/channels.proto
 INFO  | ??:??:?? 5 Installing default ChannelFile
 ERROR | ??:??:?? 5 Could not open / read /oem/oem.proto
 DEBUG | ??:??:?? 5 cleanupMeshDB purged 0 entries
 DEBUG | ??:??:?? 5 Using nodenum 0xa20a1890 
 INFO  | ??:??:?? 5 Adding node to database with 1 nodes and 233132 bytes free!
 INFO  | ??:??:?? 5 Generating new PKI keys
 DEBUG | ??:??:?? 5 Generating Curve25519 key pair...
 DEBUG | ??:??:?? 5 Number of Device Reboots: 7
 INFO  | ??:??:?? 5 Setting default channel and radio preferences!
 DEBUG | ??:??:?? 5 Expanding short PSK #1
 INFO  | ??:??:?? 5 Wanted region 0, using UNSET
 INFO  | ??:??:?? 5 Saving /prefs/config.proto
 ERROR | ??:??:?? 5 Error: can't encode protobuf io error
 ERROR | ??:??:?? 5 Can't write prefs!
 INFO  | ??:??:?? 5 Saving /prefs/channels.proto
 ERROR | ??:??:?? 5 Error: can't encode protobuf io error
 ERROR | ??:??:?? 5 Can't write prefs!
 INFO  | ??:??:?? 5 Saving /prefs/db.proto
 ERROR | ??:??:?? 5 Error: can't encode protobuf io error
 ERROR | ??:??:?? 5 Can't write prefs!
 ERROR | ??:??:?? 5 Failed to save to disk, retrying...
 INFO  | ??:??:?? 5 Saving /prefs/config.proto
 ERROR | ??:??:?? 5 Error: can't encode protobuf io error
 ERROR | ??:??:?? 5 Can't write prefs!
 INFO  | ??:??:?? 5 Saving /prefs/channels.proto
 ERROR | ??:??:?? 5 Error: can't encode protobuf io error
 ERROR | ??:??:?? 5 Can't write prefs!
 INFO  | ??:??:?? 5 Saving /prefs/db.proto
 ERROR | ??:??:?? 5 Error: can't encode protobuf io error
 ERROR | ??:??:?? 5 Can't write prefs!
```

caveman99, 2024-10-10:
try a wipe and then reflash.

liquidraver, 2024-10-10:
it was the second Erase and Flash I tried before I opened the bug report

fifieldt, 2024-10-10:
@bjpetit 's [tbeam log](https://github.com/meshtastic/firmware/issues/5020) : https://github.com/user-attachments/files/17328611/T-Beam-2.5.6-failure.txt

fifieldt, 2024-10-10:
As this affects multiple devices and prevents use of all functionality, the high-priority label felt appropriate :)

thebentern, 2024-10-10:
INFO  | ??:??:?? 6 [GPS] Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4009573c  PS      : 0x00060830  A0      : 0x80224804  A1      : 0x3ffd8740  
A2      : 0x00000002  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffcea74  
A10     : 0x3ffcea6c  A11     : 0x3f4211a3  A12     : 0x00000008  A13     : 0x3ffd8c80  
A14     : 0x3ffd8c30  A15     : 0x0000000c  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40095751  LEND    : 0x40095761  LCOUNT  : 0xfffffffd  

Backtrace: 0x40095739:0x3ffd8740 0x40224801:0x3ffd8750 0x4022b96e:0x3ffd8a60 0x4022b9aa:0x3ffd8af0 0x40149962:0x3ffd8b30 0x40149b9d:0x3ffd8b80 0x40149486:0x3ffd8bb0 0x401494a1:0x3ffd8bf0 0x40145243:0x3ffd8c20 0x401463e9:0x3ffd8ca0 0x4014688d:0x3ffd8ff0 0x40148b0e:0x3ffd9020 0x40144c12:0x3ffd9140 0x40139689:0x3ffd9160 0x400e7a1d:0x3ffd91a0
  #0  0x40095739 in strlen at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:49
  #1  0x40224801 in _svfprintf_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1528
  #2  0x4022b96e in _vsnprintf_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:70 (discriminator 4)
  #3  0x4022b9aa in vsnprintf at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:40
  #4  0x40149962 in RedirectablePrint::vprintf(char const*, char const*, __va_list_tag) at :?
  #5  0x40149b9d in RedirectablePrint::log_to_serial(char const*, char const*, __va_list_tag) at ??:?
  #6  0x40149486 in SerialConsole::log_to_serial(char const*, char const*, __va_list_tag) at ??:?
  #7  0x401494a1 in non-virtual thunk to SerialConsole::log_to_serial(char const*, char const*, __va_list_tag) at ??:?
  #8  0x40145243 in RedirectablePrint::log(char const*, char const*, ...) at ??:?
  #9  0x401463e9 in GPS::probe(int) at :?
  #10 0x4014688d in GPS::setup() at ??:?
  #11 0x40148b0e in GPS::runOnce() at ??:?
  #12 0x40144c12 in concurrency::OSThread::run() at ??:?
  #13 0x40139689 in loop() at ??:?
  #14 0x400e7a1d in loopTask(void*) at :?

caveman99, 2024-10-10:
LOG_DEBUG("Setting Baud to %i", serialSpeed); - only print in GPS::probe()

thebentern, 2024-10-10:
Fixed in latest release

closed 2024-10-10 by thebentern

VeggieVampire, 2025-01-20:
Having the same issue.

▒▒@INFO  | ??:??:?? 4

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 4 Filesystem files:
DEBUG | ??:??:?? 4  prefs (directory)
DEBUG | ??:??:?? 4    channels.proto (57 Bytes)
DEBUG | ??:??:?? 4    config.proto (170 Bytes)
DEBUG | ??:??:?? 4    module.proto (113 Bytes)
DEBUG | ??:??:?? 4    db.proto.tmp (2769 Bytes)
DEBUG | ??:??:?? 4  adafruit (directory)
DEBUG | ??:??:?? 4  bond_prph (directory)
DEBUG | ??:??:?? 4  bond_cntr (directory)
DEBUG | ??:??:?? 4 Power::lipoInit lipo sensor is not ready yet
DEBUG | ??:??:?? 4 Use analog input 5 for battery level
INFO  | ??:??:?? 4 Scan for i2c devices
DEBUG | ??:??:?? 4 Scan for I2C devices on port 1
INFO  | ??:??:?? 4 No I2C devices found
DEBUG | ??:??:?? 4 acc_info = 0
INFO  | ??:??:?? 4 S:B:9,2.5.18.89ebafc
INFO  | ??:??:?? 4 Build timestamp: 1735423672
DEBUG | ??:??:?? 4 Reset reason: 0x0
DEBUG | ??:??:?? 4 Set random seed 3384270480
INFO  | ??:??:?? 4 Init NodeDB
ERROR | ??:??:?? 4 Could not open / read /prefs/db.proto
WARN  | ??:??:?? 4 Devicestate 0 is old, discard
INFO  | ??:??:?? 4 Install default DeviceState
DEBUG | ??:??:?? 4 Initial packet id 1566294428
DEBUG | ??:??:?? 4 Partially randomized packet id 4108822941
DEBUG | ??:??:?? 4 Use nodenum 0xe717d595
INFO  | ??:??:?? 4 Load /prefs/config.proto
INFO  | ??:??:?? 4 Loaded /prefs/config.proto successfully
INFO  | ??:??:?? 4 Loaded saved config version 23
INFO  | ??:??:?? 4 Load /prefs/module.proto
INFO  | ??:??:?? 4 Loaded /prefs/module.proto successfully
INFO  | ??:??:?? 4 Loaded saved moduleConfig version 23
INFO  | ??:??:?? 4 Load /prefs/channels.proto
INFO  | ??:??:?? 4 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 4 Loaded saved channelFile version 23
ERROR | ??:??:?? 4 Could not open / read /prefs/uiconfig.proto
DEBUG | ??:??:?? 4 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 4 Use nodenum 0xe717d595
INFO  | ??:??:?? 4 Adding node to database with 1 nodes and 154840 bytes free!
DEBUG | ??:??:?? 5 Expand short PSK #1
INFO  | ??:??:?? 5 Wanted region 1, using US
DEBUG | ??:??:?? 5 Coerce telemetry to min of 30 minutes on defaults
INFO  | ??:??:?? 5 Save /prefs/db.proto
ERROR | ??:??:?? 5 LFS assert: block < lfs->cfg->block_count
ERROR | ??:??:?? 5 Trying

esev, 2025-01-20:
```
ERROR | ??:??:?? 5 LFS assert: block < lfs->cfg->block_count
ERROR | ??:??:?? 5 Trying
```

I suspect this is related to #5896 and should be fixed soon by #5900