meshtastic/firmware #5020: [Bug]: boot loop after fresh install heltec wireless tracker 1.1

megabyte2010, 2024-10-10:
### Category

Hardware Compatibility, Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.5.6

### Description

I've got a boot loop, flashing the latest firmware 2.5.6 with the online flasher, even with erasing memory. 2.5.5 works ok.

caveman99, 2024-10-10:
can you add a serial debug log?

thebentern, 2024-10-10:
Also make sure you have the correct version of the target firmware. There is a v1.0 and v1.1 of the Heltec Wireless Tracker

fifieldt, 2024-10-10:
I've building a heltec-wireless-tracker v1.1 from master much of today and it was OK...

@megabyte2010 , in case you're unfamilar with serial logs, you may obtain these by clicking the Open Serial Monitor button at the bottom of https://flasher.meshtastic.org/

bjpetit, 2024-10-10:
I'm also seeing a boot failure/loop on a T-Beam after flashing 2.5.6. I did multiple erase and install attempts with the same results. Going back to 2.5.5 works fine.
[T-Beam-2.5.6-failure.txt](https://github.com/user-attachments/files/17328611/T-Beam-2.5.6-failure.txt)

thebentern, 2024-10-10:
Looks like an existing bug report. Closing this as duplicate and tracking in 
https://github.com/meshtastic/firmware/issues/5017

closed 2024-10-10 by thebentern