meshtastic/firmware #5030: [Feature Request]: add "none" to rebroadcast mode for all* roles as a configurable option

b8b8, 2024-10-11:
### Platform

Cross-Platform

### Description

Right now the only role that truly doesnt participate in meshing is client_mute. 

I believe there should be a "no rebroadcast" option added to the drop-down list so that at least sensor and tracker role can use this.

A scenario where you have a room full of powered sensors or a single powered (always awake) weather station would benefit from the higher telemetry priority in the sensor role while still not wanting to contribute to meshing other packets and cause congestion. Currently if there is high channel utilization a client_mute device would stop sending or scale telemetry. This would mean a weather station would never be able to pass on data to the rest of the busy mesh. 

The same could can be said for always powered (car?) trackers that prioritize position data but we still dont want to participate in meshing. 

Client_Hidden_**mute** would also be configurable if this option was added.

thebentern, 2024-10-11:
I like this idea. Prevents creation of unnecessary roles.

b8b8, 2024-10-11:
yup, and i really like the idea of options to make everything "quieter" on the mesh.

fifieldt, 2024-10-11:
Error if it's enabled on router or repeater roles?

b8b8, 2024-10-11:
> Error if it's enabled on router or repeater roles?

yes that makes sense or at least greyed out on the two apps.

caveman99, 2024-10-11:
@thebentern added this before i saw you took this :-) https://github.com/meshtastic/protobufs/pull/599

PR #5040 (2024-10-12): Implement rebroadcast mode NONE
Closes #5030
closed 2024-10-12 by thebentern