meshtastic/firmware #5032: [Bug]: Segmentation fault with `armhf` binary on Raspberry Pi 3B

roens, 2024-10-11:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.5.4.8d288d5

### Description

I'm attempting to run `meshtasticd` on a Raspberry Pi 3 Model B Rev 1.2. Attached to the RPi is an [Uputronics LoRa Expansion HAT](), with two RFM95 radio modules. For several releases now (`2.3.13.83f5ba0`, `2.3.14.64531fa`, `2.4.2.5b45303`), attempting to run the installed binary results in:

```shell
$ meshtasticd
Portduino is starting, HWID=1, VFS root at /root/.portduino/default
Setting up Meshtastic on Portduino...
Using /etc/meshtasticd/config.yaml as config file
Segmentation fault
```

My guess is that config has little to do with this segfault as it fails right quick. But what I've added to `/etc/meshtasticd/config.yaml` is:

```yaml
Lora:
  Module: RF95  # Uputronics LoRa Expansion Board
  CS: 24   #CE0 CS/NSS
  IRQ: 32  #CE0 DIO0
```

I've looked a number of times for any Issues or Discussion about others experiencing segfaults with `meshtasticd`, but come up with little. Perhaps I'm missing something, or perhaps this genuinely is a novel issue?

### Relevant log output

```Shell
$ type -f meshtasticd
meshtasticd is /usr/sbin/meshtasticd

$ file /usr/sbin/meshtasticd 
/usr/sbin/meshtasticd: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (GNU/Linux), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, BuildID[sha1]=452f71bf68b507d331d6b38f7063880a183d5391, for GNU/Linux 3.2.0, with debug_info, not stripped

$ uname -a
Linux LoRa-Gate 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

$ ldd /usr/sbin/meshtasticd
	linux-vdso.so.1 (0x7ee20000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x743a0000)
	libgpiod.so.2 => /lib/arm-linux-gnueabihf/libgpiod.so.2 (0x74370000)
	libyaml-cpp.so.0.7 => /lib/arm-linux-gnueabihf/libyaml-cpp.so.0.7 (0x74310000)
	libulfius.so.2.7 => /lib/arm-linux-gnueabihf/libulfius.so.2.7 (0x742d0000)
	liborcania.so.2.3 => /lib/arm-linux-gnueabihf/liborcania.so.2.3 (0x742a0000)
	libcrypto.so.3 => /lib/arm-linux-gnueabihf/libcrypto.so.3 (0x73f50000)
	libstdc++.so.6 => /lib/arm-linux-gnueabihf/libstdc++.so.6 (0x73d80000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x73d39000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x73d00000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x73b87000)
	/lib/ld-linux-armhf.so.3 (0x76fb2000)
	libgnutls.so.30 => /lib/arm-linux-gnueabihf/libgnutls.so.30 (0x73905000)
	libmicrohttpd.so.12 => /lib/arm-linux-gnueabihf/libmicrohttpd.so.12 (0x738e1000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x738a0000)
	libcurl-gnutls.so.4 => /lib/arm-linux-gnueabihf/libcurl-gnutls.so.4 (0x73813000)
	libjansson.so.4 => /lib/arm-linux-gnueabihf/libjansson.so.4 (0x737f0000)
	libyder.so.2.0 => /lib/arm-linux-gnueabihf/libyder.so.2.0 (0x737c0000)
	libatomic.so.1 => /lib/arm-linux-gnueabihf/libatomic.so.1 (0x73790000)
	libp11-kit.so.0 => /lib/arm-linux-gnueabihf/libp11-kit.so.0 (0x73670000)
	libidn2.so.0 => /lib/arm-linux-gnueabihf/libidn2.so.0 (0x73641000)
	libunistring.so.2 => /lib/arm-linux-gnueabihf/libunistring.so.2 (0x73480000)
	libtasn1.so.6 => /lib/arm-linux-gnueabihf/libtasn1.so.6 (0x73450000)
	libnettle.so.8 => /lib/arm-linux-gnueabihf/libnettle.so.8 (0x733e0000)
	libhogweed.so.6 => /lib/arm-linux-gnueabihf/libhogweed.so.6 (0x73380000)
	libgmp.so.10 => /lib/arm-linux-gnueabihf/libgmp.so.10 (0x732f0000)
	libnghttp2.so.14 => /lib/arm-linux-gnueabihf/libnghttp2.so.14 (0x732ca000)
	librtmp.so.1 => /lib/arm-linux-gnueabihf/librtmp.so.1 (0x732a0000)
	libssh2.so.1 => /lib/arm-linux-gnueabihf/libssh2.so.1 (0x73250000)
	libpsl.so.5 => /lib/arm-linux-gnueabihf/libpsl.so.5 (0x7323d000)
	libgssapi_krb5.so.2 => /lib/arm-linux-gnueabihf/libgssapi_krb5.so.2 (0x731fb000)
	libldap-2.5.so.0 => /lib/arm-linux-gnueabihf/libldap-2.5.so.0 (0x731aa000)
	liblber-2.5.so.0 => /lib/arm-linux-gnueabihf/liblber-2.5.so.0 (0x74393000)
	libzstd.so.1 => /lib/arm-linux-gnueabihf/libzstd.so.1 (0x7310b000)
	libbrotlidec.so.1 => /lib/arm-linux-gnueabihf/libbrotlidec.so.1 (0x730e0000)
	libsystemd.so.0 => /lib/arm-linux-gnueabihf/libsystemd.so.0 (0x73026000)
	libffi.so.8 => /lib/arm-linux-gnueabihf/libffi.so.8 (0x73000000)
	libkrb5.so.3 => /lib/arm-linux-gnueabihf/libkrb5.so.3 (0x72f52000)
	libk5crypto.so.3 => /lib/arm-linux-gnueabihf/libk5crypto.so.3 (0x72f28000)
	libcom_err.so.2 => /lib/arm-linux-gnueabihf/libcom_err.so.2 (0x7436c000)
	libkrb5support.so.0 => /lib/arm-linux-gnueabihf/libkrb5support.so.0 (0x74305000)
	libsasl2.so.2 => /lib/arm-linux-gnueabihf/libsasl2.so.2 (0x72ef0000)
	libbrotlicommon.so.1 => /lib/arm-linux-gnueabihf/libbrotlicommon.so.1 (0x72eb0000)
	libcap.so.2 => /lib/arm-linux-gnueabihf/libcap.so.2 (0x742c5000)
	libgcrypt.so.20 => /lib/arm-linux-gnueabihf/libgcrypt.so.20 (0x72da0000)
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x72d78000)
	liblz4.so.1 => /lib/arm-linux-gnueabihf/liblz4.so.1 (0x72d58000)
	libkeyutils.so.1 => /lib/arm-linux-gnueabihf/libkeyutils.so.1 (0x72d30000)
	libresolv.so.2 => /lib/arm-linux-gnueabihf/libresolv.so.2 (0x738d2000)
	libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0x72cf0000)
```

jp-bennett, 2024-10-12:
There's a known issue with the armhf binaries for the Pi. The arm64 images should be working without the segfault.

jp-bennett, 2024-10-12:
We'll track this with #4978

closed 2024-10-12 by jp-bennett

roens, 2024-10-13:
> There's a known issue with the armhf binaries for the Pi. The arm64 images should be working without the segfault.

Thanks @jp-bennett… I did try installing `meshtasticd_2.5.5.e182ae7_arm64.deb`, but got complaints about unavailable dependencies. But then that's already been discussed 😄 #3563

Looking forward to future releases, thanks for your work!