meshtastic/firmware #5035: [Bug]: Web client not bundled with 2.5.5 release served via Web Flasher

todd-herbert, 2024-10-11:
### Category

WiFi

### Hardware

Not Applicable

### Firmware Version

2.5.5.e182ae7

### Description

Discord user tenhorac reports that the 2.5.5 build server by the web flasher does not include the web client, instead showing the `Web server is running.` message. I can confirm that this is also the case with the 2.5.5 build served for Wireless Paper.

Versions 2.5.4 and 2.5.6 are both working correctly; it seems that only 2.5.5 was affected.

### Relevant log output

```Shell
[HTTPS:I] New connection. Socket FID=52
[HTTPS:I] Request: GET / (FID=52)
[169432][E][vfs_api.cpp:105] open(): /littlefs/static/index.html does not exist, no permits for creation
[169443][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
[169454][E][vfs_api.cpp:105] open(): /littlefs/static/index.html.gz does not exist, no permits for creation
WARN  | 12:40:47 169 [WebServerThread] File not available - /static/index.html.gz
```

thebentern, 2024-10-11:
Something perhaps went wrong with the pulling of web build.tar from assets. Since this is resolved upstream in the latest version, ok if I close this and leave an admonition on the 2.5.5 release?

caveman99, 2024-10-11:
i think we need to fail the build when the pull does not succeed.

thebentern, 2024-10-11:
> i think we need to fail the build when the pull does not succeed.

I agree. We need to take a look and see if there are any obvious failures in the build logs for the offending release

closed 2024-10-14 by todd-herbert

tomaszrondio, 2025-02-25:
I am affected by this bug using 2.5.20

fifieldt, 2025-02-25:
@tomaszrondio , please see: https://github.com/meshtastic/firmware/discussions/5381

tomaszrondio, 2025-02-25:
Is it even possible to manage this node over wifi?

fifieldt, 2025-02-25:
Yup! Just flash the firmware at flasher.meshtastic.org, tick the full erase and tick the website button. It'll all work just like it use to.

tomaszrondio, 2025-02-25:
Beautiful! Thank you

Phobos-7, 2025-08-30:
Still doesn't work on T-Beam Supreme

Web server is running.
The content you are looking for can't be found. Please see: [FAQ](https://meshtastic.org/docs/software/web-client/).
[admin](http://meshtastic.local/admin)