meshtastic/firmware #5037: [Bug]: T-Beam remote management over LoRa network is not working!

velasco79, 2024-10-11:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.5.5.e182ae7 Beta

### Description

T-Beam does not allow management through the LoRa network even with the public key of the administrator device registered.

### Relevant log output

```Shell
[0m [34m DEBUG [0m| 15:06:55 666 [RadioIf] [34m Lora RX (id=0x17d1c84d fr=0x50 to=0xc8, WantAck=1, HopLim=3 Ch=0x0 encrypted rxSNR=11.25 rxRSSI=-53 hopStart=3)
[0m [34m DEBUG [0m| 15:06:55 666 [RadioIf] [34m Packet received : 600ms
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Add packet record (id=0x17d1c84d fr=0x50 to=0xc8, WantAck=1, HopLim=3 Ch=0x0 encrypted rxSNR=11.25 rxRSSI=-53 hopStart=3)
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Attempting PKI decryption
[0m [32m INFO  [0m| 15:06:55 666 [Router] [32m Random nonce value: 2098560397
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Generating DH with remote pubkey:  7d fd 25 3b 69 5a 66 0e 14 c4 93 17 aa 52 37 7d 9b 4a 10 8a ba e3 3a 44 8d b0 9f 4a fd ae 64 10
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m And local pubkey:  78 76 d1 da 7f ff 36 55 36 8e 18 75 f7 7a 72 05 8e 7e e8 36 30 8e 8a 7a 5a b7 2b 0e 34 fc 11 72
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Attempting decrypt using nonce:  4d c8 d1 17 8d 7d 15 7d 50 a8 3c 43 00
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Attempting decrypt using shared_key starting with:  2e df 4b 4a 62 85 f9 03
[0m [33m WARN  [0m| 15:06:55 666 [Router] [33m PKC decrypt attempted but failed!
[0m [33m WARN  [0m| 15:06:55 666 [Router] [33m No suitable channel found for decoding, hash was 0x0!
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m packet decoding failed or skipped (no PSK?) (id=0x17d1c84d fr=0x50 to=0xc8, WantAck=1, HopLim=3 Ch=0x0 encrypted rxtime=1728659215 rxSNR=11.25 rxRSSI=-53 hopS
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Module 'routing' wantsPacket=1
[0m [32m INFO  [0m| 15:06:55 666 [Router] [32m Received routing from=0x433ca850, id=0x17d1c84d, portnum=-987430879, payloadlen=1856
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Routing sniffing (id=0x17d1c84d fr=0x50 to=0xc8, WantAck=1, HopLim=3 Ch=0x0 encrypted rxtime=1728659215 rxSNR=11.25 rxRSSI=-53 hopStart=3)
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Partially randomized packet id 2996276590
[0m [33m WARN  [0m| 15:06:55 666 [Router] [33m Alloc an err=6,to=0x433ca850,idFrom=0x17d1c84d,id=0xb2978d6e
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Add packet record (id=0xb2978d6e fr=0xc8 to=0x50, WantAck=0, HopLim=2 Ch=0x0 Portnum=5 requestId=17d1c84d rxtime=1728659215 priority=120)
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Expanding short PSK #1
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Using AES128 key!
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m enqueuing for send (id=0xb2978d6e fr=0xc8 to=0x50, WantAck=0, HopLim=2 Ch=0x8 encrypted rxtime=1728659215 hopStart=2 priority=120)
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m txGood=6,txRelay=0,rxGood=7,rxBad=0
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Delivering rx packet (id=0x17d1c84d fr=0x50 to=0xc8, WantAck=1, HopLim=3 Ch=0x0 encrypted rxtime=1728659215 rxSNR=11.25 rxRSSI=-53 hopStart=3)
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Forwarding to phone (id=0x17d1c84d fr=0x50 to=0xc8, WantAck=1, HopLim=3 Ch=0x0 encrypted rxtime=1728659215 rxSNR=11.25 rxRSSI=-53 hopStart=3)
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Attempting PKI decryption
[0m [32m INFO  [0m| 15:06:55 666 [Router] [32m Random nonce value: 2098560397
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Generating DH with remote pubkey:  7d fd 25 3b 69 5a 66 0e 14 c4 93 17 aa 52 37 7d 9b 4a 10 8a ba e3 3a 44 8d b0 9f 4a fd ae 64 10
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m And local pubkey:  78 76 d1 da 7f ff 36 55 36 8e 18 75 f7 7a 72 05 8e 7e e8 36 30 8e 8a 7a 5a b7 2b 0e 34 fc 11 72
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Attempting decrypt using nonce:  4d c8 d1 17 8d 7d 15 7d 50 a8 3c 43 00
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Attempting decrypt using shared_key starting with:  2e df 4b 4a 62 85 f9 03
[0m [33m WARN  [0m| 15:06:55 666 [Router] [33m PKC decrypt attempted but failed!
[0m [33m WARN  [0m| 15:06:55 666 [Router] [33m No suitable channel found for decoding, hash was 0x0!
[0m [34m DEBUG [0m| 15:06:55 666 [Router] [34m Module 'routing' considered
[0m [32m INFO  [0m| 15:06:55 666 [32m Telling client we have new packets 18
[0m [32m INFO  [0m| 15:06:55 666 [SerialConsole] [32m getFromRadio=STATE_SEND_PACKETS
[0m [34m DEBUG [0m| 15:06:55 666 [SerialConsole] [34m phone downloaded packet (id=0x17d1c84d fr=0x50 to=0xc8, WantAck=1, HopLim=3 Ch=0x0 encrypted rxtime=1728659215 rxSNR=11.25 rxRSSI=-53 hopStart=3)
[0m �� OMP�<C�+��*!%�@�
��\�m��KE�AL���)1��}}5M��=?	gE  4AHP`���������x[34m DEBUG [0m| 15:06:55 666 [RadioIf] [34m Starting low level send (id=0xb2978d6e fr=0xc8 to=0x50, WantAck=0, HopLim=2 Ch=0x8 encrypted rxtime=1728659215 hopStart=2 priority=120)
[0m [34m DEBUG [0m| 15:06:55 667 [RadioIf] [34m Packet transmitted : 452ms
[0m [34m DEBUG [0m| 15:06:56 667 [RadioIf] [34m Completed sending (id=0xb2978d6e fr=0xc8 to=0x50, WantAck=0, HopLim=2 Ch=0x8 encrypted rxtime=1728659215 hopStart=2 priority=120)
[0m [34m DEBUG [0m| 15:07:05 676 [Power] [34m Battery: usbPower=1, isCharging=1, batMv=4205, batPct=100
[0m [32m INFO  [0m| 15:07:25 696 [DeviceTelemetryModule] [32m (Sending): air_util_tx=0.107694, channel_utilization=1.753333, battery_level=101, voltage=4.205000, uptime=696
[0m [34m DEBUG [0m| 15:07:25 696 [DeviceTelemetryModule] [34m Partially randomized packet id 1688827247
[0m [34m DEBUG [0m| 15:07:25 696 [DeviceTelemetryModule] [34m updateTelemetry LOCAL
[0m [34m DEBUG [0m| 15:07:25 696 [DeviceTelemetryModule] [34m Node status update: 2 online, 2 total
[0m [32m INFO  [0m| 15:07:25 696 [DeviceTelemetryModule] [32m Sending packet to phone
[0m [32m INFO  [0m| 15:07:25 696 [32m Telling client we have new packets 19
[0m [34m DEBUG [0m| 15:07:25 696 [Power] [34m Battery: usbPower=1, isCharging=1, batMv=4204, batPct=100
[0m [32m INFO  [0m| 15:07:25 696 [SerialConsole] [32m getFromRadio=STATE_SEND_PACKETS
[0m [34m DEBUG [0m| 15:07:25 696 [SerialConsole] [34m phone downloaded packet (id=0x64a9756f fr=0xc8 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1728659245 priority=10)
[0m �� ;9�+������"C-?	ge\��@;m�?%��=(�5ou�d=-?	gHX
[34m DEBUG [0m| 15:07:45 716 [Power] [34m Battery: usbPower=1, isCharging=1, batMv=4204, batPct=100
[0m [34m DEBUG [0m| 15:08:05 736 [Power] [34m Battery: usbPower=1, isCharging=1, batMv=4205, batPct=100
[0m
```

jp-bennett, 2024-10-12:
Can you send DMs at all between the two nodes? `PKC decrypt attempted but failed!` would indicate probably not. Two things. This is the log from the receiving node. Can you do the same thing, but capture the log from the other node? The one actually sending the packets? Then, you can try telling each node to forget the other one, and letting them rediscover each other. This should pull the public keys again, and hopefully get things working.

velasco79, 2024-10-12:
Good evening, thank you very much for your attention.

What you advised me to do worked!
I asked to forget the T-beam and when it appeared again, the remote
management worked.

Thank you very much.

closed 2024-10-13 by jp-bennett

mike6789, 2024-10-22:
This is really irritating, why is the issue closed?

I was able to exchange messages between client and router, but not the --request-telemetry command... I had to reset the nodedb on both to get it working

jp-bennett, 2024-10-22:
> This is really irritating, why is the issue closed?

Because:

> What you advised me to do worked!

If you find a case where it's not just a key mismatch, open a new issue for it.

mike6789, 2024-10-22:
> > This is really irritating, why is the issue closed?
> 
> Because:
> 
> > What you advised me to do worked!
> 
> If you find a case where it's not just a key mismatch, open a new issue for it.

In the case I described and just experienced, regular messages are properly decrypted by the router, but the telemetry message isn't! Doesn't that mean the key was not mis-matched? Thanks for replying ... this was frustrating me for hours!!

jp-bennett, 2024-10-22:
> In the case I described and just experienced, regular text messages are properly decrypted by the router, but the telemetry message isn't! Doesn't that mean the key was not mis-matched? Thanks for replying ... this was frustrating me for hours!!

When you tested with messages, were they Direct Messages, or just channel broadcasts? Happen to have any debug logs?

mike6789, 2024-10-22:
> When you tested with messages, were they Direct Messages, or just channel broadcasts? Happen to have any debug logs?

Just broadcasts -- I've lost all the logs for now. I can try to reproduce it again but I don't know how I got into that state