meshtastic/firmware #5048: [Bug]: Enabling Wi-Fi makes T-LoRas unstable

IZ1IVA, 2024-10-13:
### Category

WiFi

### Hardware

T-Lora v2 1.6

### Firmware Version

2.4.3

### Description

Ever since I've upgraded my T-LoRas to 2.4.3 I'm experiencing issues in keeping the nodes up and running. After a successful boot, the nodes seem to be freezing up at random, both in terms of when happens and for how long. Yes, they seem to come alive again on their own.
I'm observing the same issue with firmware 2.5.6.

T-Beams, on the other hand, are working fine with the same configuration.

pdxlocations, 2024-10-14:
Have you tried a full erase and reinstall? I’m running that board without issues.

IZ1IVA, 2024-10-14:
> Have you tried a full erase and reinstall? I’m running that board without issues.

I did a full install using the _device-install_ script (not the _device-upgrade_), over a previous firmware.

Are you suggesting a `meshtastic --set factory_reset true` from the CLI? Is that more thorough than the erasing routine in the script?

pdxlocations, 2024-10-14:
> I did a full install using the _device-install_ script (not the _device-upgrade_), over a previous firmware.

Yes this is what I'm referring to.  Since I'm having no issues with the firmware on the same board, it's likely a hardware issue, unless you've changed settings.  WiFi can be finicky as well.  You may need to move the board closer to your router.  You could try another power source, USB cable, etc.  If you have another V2.1-1.6, try replicating with that.

Another note, I've had these boards appear unresponsive when connected to MQTT.  I sometimes can't connect with a client (unless using serial) but they are chugging away in the background feeding my broker with mesh data.

IZ1IVA, 2024-10-15:
> Yes this is what I'm referring to. Since I'm having no issues with the firmware on the same board, it's likely a hardware issue, unless you've changed settings. WiFi can be finicky as well. You may [...]

I'm currently testing this bug on 3 T-LoRas of different vintage, plus a local friend is experiencing the same issue on his T-LoRa.
It seems they are all working fine with 2.3 firmware, while anything newer sooner or later makes the board freeze just the Wi-Fi part while the LoRa part keeps on working, or puts the whole board into (deep?) sleep.
Another unusual behaviour is the Android app sometimes connecting via IP only on a second attempt.

We are using default settings save for:

-  MediumFast channel (all)
-  fixed position (all)
-  Wi-Fi (all)
-  environment telemetry (some)
- MQTT (some)

All these parameters are unchanged from the past , same as the Wi-Fi routers.

> Another note, I've had these boards appear unresponsive when connected to MQTT. I sometimes can't connect with a client (unless using serial) but they are chugging away in the background feeding my broker with mesh data.

That's interesting, thank you for pointing it out. It might play a role, however not all our nodes have MQTT configured, plus we are using our own MQTT server, which should be managing far less traffic than the official one.

Cheers

IZ1IVA, 2024-10-19:
I'm doing some more testing and it seems the same T-LoRa boards are not having these issues when running firmware 2.3.15.

CamFlyerCH, 2024-10-20:
I always had issues with these boards when enabling Wifi. I think while sending via Wifi, the power delivery is insufficient and the chip has a "brown out" and resets. I found a long time ago an article suggesting to add a capacitor on the boards power rail. My only one of these boards in use with Wifi has an old phone charger that supplies around 5.2v to the board. 
If you have the "brown out" issue, you can see it via USB serial in the messages of the crash and reboot from the device itself, not the Metastatic software.

IZ1IVA, 2024-10-20:
> I always had issues with these boards when enabling Wifi. I think while sending via Wifi, the power delivery is insufficient and the chip has a "brown out" and resets. I found a long time ago an article suggesting to add a capacitor on the boards power rail. 

That's interesting, thanks for sharing your experience!

I've looked into this, but I'm not seeing any browout by looking at the serial console. They are both powered via godd quality USB cables hooked up to a computer, so I'd rule out power issues.
To be fair, the local boards are not spontaneously rebooting anymore, what gives?

I need to retrieve the remote node for further testing.

IZ1IVA, 2024-10-31:
After further testing, I can confirm that disabling Wi-Fi seems to solve the issue, hence something must have changed between firmware 2.3 and 2.4.

kacian, 2025-04-04:
I have the same problem on all tloras i use.

Logs when it rebooting during app connect:
```
INFO  | 16:43:25 6595 [ApiServer] Force close previous TCP connection
DEBUG | 16:43:25 6595 [ApiServer] PhoneAPI::close()
INFO  | 16:43:25 6595 [ApiServer] Incoming API connection
INFO  | 16:43:25 6595 [ApiServer] Incoming wifi connection
INFO  | 16:43:25 6595 [ServerAPI] Client wants config, nonce=34

abort() was called at PC 0x401eaebf on core 1

Backtrace: 0x4008414d:0x3ffd7700 0x400976a1:0x3ffd7720 0x4009d4d5:0x3ffd7740 0x401eaebf:0x3ffd77c0 0x401eaf06:0x3ffd77e0 0x401eb23b:0x3ffd7800 0x401eb19a:0x3ffd7820 0x40141ab5:0x3ffd7840 0x401418ca:0x3ffd79a0 0x4012f7aa:0x3ffd7b00 0x4012640e:0x3ffd7d50 0x40131c3b:0x30

##############

INFO  | 00:04:55 2263 [ApiServer] Incoming API connection
INFO  | 00:04:55 2263 [ApiServer] Incoming wifi connection
INFO  | 00:04:55 2263 [ServerAPI] Client wants config, nonce=6
[2264098][E][vfs_api.cpp:301] VFSFileImpl(): fopen(/littlefs/prefs/module.proto) failed

abort() was called at PC 0x401ecf9f on core 1

Backtrace: 0x4008414d:0x3ffd78c0 0x400976a1:0x3ffd78e0 0x4009d4d5:0x3ffd7900 0x401ecf9f:0x3ffd7980 0x401ecfe6:0x3ffd79a0 0x401ed32b:0x3ffd79c0 0x401ed28a:0x3ffd79e0 0x40143985:0x3ffd7a00 0x4012fada:0x3ffd7b60 0x4012718a:0x3ffd7db0 0x40133107:0x3ffd7dd0 0x40138899:0x3ffd7df0 0x40140b06:0x3ffd7e10 0x40135145:0x3ffd70

###############

INFO  | 15:50:31 3104 [ApiServer] Incoming API connection
INFO  | 15:50:31 3104 [ApiServer] Incoming wifi connection
INFO  | 15:50:31 3104 [ServerAPI] Client wants config, nonce=46

abort() was called at PC 0x401ecf9f on core 1

Backtrace: 0x4008414d:0x3ffd7760 0x400976a1:0x3ffd7780 0x4009d4d5:0x3ffd77a0 0x401ecf9f:0x3ffd7820 0x401ecfe6:0x3ffd7840 0x401ed32b:0x3ffd7860 0x401ed28a:0x3ffd7880 0x40143ab1:0x3ffd78a0 0x401438c6:0x3ffd7a00 0x4012fada:0x3ffd7b60 0x4012718a:0x3ffd7db0 0x40133107:0x30

####################

INFO  | 19:59:57 1546 [ApiServer] Incoming API connection
INFO  | 19:59:57 1546 [ApiServer] Incoming wifi connection
INFO  | 19:59:57 1546 [ServerAPI] Client wants config, nonce=49

abort() was called at PC 0x401ecf9f on core 1

Backtrace: 0x4008414d:0x3ffd7760 0x400976a1:0x3ffd7780 0x4009d4d5:0x3ffd77a0 0x401ecf9f:0x3ffd7820 0x401ecfe6:0x3ffd7840 0x401ed32b:0x3ffd7860 0x401ed28a:0x3ffd7880 0x40143ab1:0x3ffd78a0 0x401438c6:0x3ffd7a00 0x4012fada:0x3ffd7b60 0x4012718a:0x3ffd7db0 0x40133107:0x30

####################

INFO  | 21:42:37 29028 [ApiServer] Force close previous TCP connection
DEBUG | 21:42:37 29028 [ApiServer] PhoneAPI::close()
INFO  | 21:42:37 29028 [ApiServer] Incoming API connection
INFO  | 21:42:37 29028 [ApiServer] Incoming wifi connection
INFO  | 21:42:37 29028 [ServerAPI] Client wants config, nonce=101

abort() was called at PC 0x401eb42b on core 1

Backtrace: 0x4008414d:0x3ffd7850 0x400976a1:0x3ffd7870 0x4009d4d5:0x3ffd7890 0x401eb42b:0x3ffd7910 0x401eb472:0x3ffd7930 0x401eb7b7:0x3ffd7950 0x401eb716:0x3ffd7970 0x40141f51:0x3ffd7990 0x4012f2fa:0x3ffd7af0 0x40126532:0x3ffd7d40 0x4013216f:0x3ffd7d60 0x40137d49:0x30

#####################

[HTTPS:I] New connection. Socket FID=54
[HTTPS:I] Request: OPTIONS /api/v1/toradio (FID=54)
DEBUG | 22:32:34 2070 [WebServer] webAPI handleAPIv1ToRadio
[HTTPS:I] Request: PUT /api/v1/toradio (FID=54)
DEBUG | 22:32:34 2070 [WebServer] webAPI handleAPIv1ToRadio
DEBUG | 22:32:34 2070 [WebServer] Received 6 bytes from PUT request
INFO  | 22:32:34 2070 [WebServer] Client wants config, nonce=877566593

abort() was called at PC 0x401ea5c3 on core 1

Backtrace: 0x4008414d:0x3ffd72a0 0x400976a1:0x3ffd72c0 0x4009d4d5:0x3ffd72e0 0x401ea5c3:0x3ffd7360 0x401ea60a:0x3ffd7380 0x401ea93f:0x3ffd73a0 0x401ea89e:0x3ffd73c0 0x401412b5:0x3ffd73e0 0x401410ca:0x3ffd7540 0x4012f526:0x3ffd76a0 0x40126861:0x3ffd78f0 0x400f500f:0x30
[... 155 lines cut ...]
Backtrace: 0x40084149:0x3ffd77c0 0x4009769d:0x3ffd77e0 0x4009d4d1:0x3ffd7800 0x401ecf4b:0x3ffd7880 0x401ecf92:0x3ffd78a0 0x401ed2d7:0x3ffd78c0 0x401ed236:0x3ffd78e0 0x40143a5d:0x3ffd7900 0x40143872:0x3ffd7a60 0x4012fb36:0x3ffd7bc0 0x401271ba:0x3ffd7e10 0x401330d7:0x30

#####################

INFO  | 19:56:23 2895 [ApiServer] Force close previous TCP connection
DEBUG | 19:56:23 2895 [ApiServer] PhoneAPI::close()
INFO  | 19:56:23 2895 [ApiServer] Incoming API connection
INFO  | 19:56:23 2895 [ApiServer] Incoming wifi connection
INFO  | 19:56:23 2895 [ServerAPI] Client wants config, nonce=8

abort() was called at PC 0x401ecf4b on core 1

Backtrace: 0x40084149:0x3ffd7920 0x4009769d:0x3ffd7940 0x4009d4d1:0x3ffd7960 0x401ecf4b:0x3ffd79e0 0x401ecf92:0x3ffd7a00 0x401ed2d7:0x3ffd7a20 0x401ed236:0x3ffd7a40 0x40143931:0x3ffd7a60 0x4012fb36:0x3ffd7bc0 0x401271ba:0x3ffd7e10 0x401330d7:0x3ffd7e30 0x401388a9:0x30

####################

DEBUG | 12:36:18 18936 [WebServer] webAPI handleAPIv1ToRadio
DEBUG | 12:36:18 18936 [WebServer] Received 6 bytes from PUT request
INFO  | 12:36:18 18936 [WebServer] Client wants config, nonce=683959778

abort() was called at PC 0x401eb3e7 on core 1

Backtrace: 0x4008414d:0x3ffd7380 0x400976a1:0x3ffd73a0 0x4009d4d5:0x3ffd73c0 0x401eb3e7:0x3ffd7440 0x401eb42e:0x3ffd7460 0x401eb773:0x3ffd7480 0x401eb6d2:0x3ffd74a0 0x40141f11:0x3ffd74c0 0x4012f302:0x3ffd7620 0x40126fad:0x3ffd7870 0x400f55af:0x3ffd7ad0 0x400f7c16:0x30

```

kacian, 2025-04-04:
But #6493 does not fix rebooting on connection from the app via wifi.

kacian, 2025-04-04:
Please reopen cause the problem is not solved.

itxpress, 2025-04-19:
I'm seeing similar wireless issues with my TLora on v 2.5.6. It'll work fine, then drop off the network. It's still running most of the time,  I can send messages to it. I can often remote admin to it from my TBeam and get it to reboot. I can send messages to it. But it's going up and down on wifi a lot. 

I'm actually having a somewhat similar issue with my Heltec v3 MQTT node. It's randomly disappearing off the network and randomly rebooting. Not sure if it's a memory leak due to high traffic, or what. The Heltec I have subscribed to msh/US, while the TLora I limit to msh/US/NC  I just turned MQTT off on the Heltec to see how long it stays running. 

The TBeam is linked to a local MQTT server and has been rock solid on WiFI. All these nodes are on 2.6.4/2.6.5

allsupj, 2025-05-12:
Same issue here with the T3S3 model. WiFi connection just stops but all else appears to keep working. Reboot restores Wifi

PR #6493 (2025-04-04): Revert "Try-fix ESP32 wifi disconnects"
Closes #5048
closed 2025-07-07 by github-actions[bot]